don/BluetoothSerial #98: The connect method called connectFailure but is connected

luckebar, 2015-01-01:
The connect method called connectFailure but is connected.

The error message is: Unable to connect to device, but after a few seconds posts by Arduino start arriving. Until recently it worked well. What could have happened? Were updated libraries? It may be my mistake?

Thank You

don, 2015-01-02:
Are you seeing this on iOS or Android? What version of Cordova are you using? I did see some problems with Cordova 4.1 and multiple callbacks being called. I downgraded to 4.0 since I didn't have time to look into the details.

luckebar, 2015-01-02:
I use intel XDK cordova 3.5 I think.
I'm testing with Android. I tried again with a version of a few days ago and the connection works fine. The .js file with all the functions for the bluetooth is the same. Maybe it faults some other javascript function or I used some or variable name reserved for libraries bluetooth. It could be one of these causes?

don, 2015-02-20:
This should be fixed in the latest code. If not re-open this issue.

closed 2015-02-20 by don