don/BluetoothSerial #101: Increase timeout on bluetoothSerial.list()?

mrtomhoward, 2015-01-22:
For the BLE device I'm working with, the engineer wants to reduce the advertising rate to 5 seconds to minimise power consumption.

It seems this is too slow for the bluetoothSerial.list() method.

Is this configurable somewhere?

don, 2015-01-23:
@mrtomhoward the scan time is not configurable yet.

You'll need to edit this in 2 places: the call to [scan](https://github.com/don/BluetoothSerial/blob/fdca9cf6d3bc0bfeebadec62c335934952b8440f/src/ios/MEGBluetoothSerial.m#L155) and the [timer](https://github.com/don/BluetoothSerial/blob/fdca9cf6d3bc0bfeebadec62c335934952b8440f/src/ios/MEGBluetoothSerial.m#L157).

I need to think about the API change. Scan time could passed as additional parameter to `list` or perhaps a plugin configuration that can be adjusted.

mrtomhoward, 2015-01-28:
@don That works nicely, many thanks for the quick reply.

closed 2015-01-28 by mrtomhoward