don/BluetoothSerial #102: bluetoothSerial.connect() not working for Android 5.0.1?

rjwhite, 2015-01-24:
My App that was working fine for Android 4.4.4 no longer works for Android 5.0.1
I can't be sure if this is a cordova problem of insisting on using API-19 (4.4.2)  when I want it to use API-21 (5.0.1) or if the problem is here in this plugin.  It fails in bluetoothSerial.connect()
This is on my LG Nexus 4 that just got upgraded to 5.0.1.
The phone can pair to my Arduino Adafruit EZ-Link shield, the App can verify Bluetooth is enabled, get the Bluetooth device list and successfully search for the device name and get the MAC, but it can no longer connect.
I also tried bluetoothSerial.connectInsecure() with the same results.

don, 2015-01-24:
You should continue to use API level 19 with Cordova even when deploying to your Nexus 4 running 5.0.1 (API 21)

rjwhite, 2015-01-24:
ok, thanx Don!  So.... can you tell me if there is an issue then with BluetoothSerial for 5.0.1 or is my problem something else?

don, 2015-01-24:
I haven't seen problems with 5.0. See if you're getting errors in `adb logcat` or in the webview (use chrome://inspect). I'll try and test a similar setup.

rjwhite, 2015-01-24:
I don't totally trust my cordova install right now.  To go back to API 19 means I need to reinstall everything from scratch again or else nothing at all works.  Right now, the mish-mash of crap I have at least finds the pics and CSS and it *looks* ok, except I don't get a connect.  However, amongst the infinite output from logcat I see:

E/BluetoothSerialService(18603): java.io.IOException: read failed, socket might closed or timeout, read ret: -1
I/BluetoothSerialService(18603): Trying fallback...
W/BluetoothAdapter(18603): getBluetoothService() called with no BluetoothManagerCallback
E/BluetoothSerialService(18603): Couldn't establish a Bluetooth connection.

rjwhite, 2015-01-24:
ok, I just rebuilt everything from scratch again.  This is what I see in logcat:
...
D/BluetoothSerial(24567): action = isEnabled
D/BluetoothSerial(24567): action = list
D/BluetoothSerial(24567): action = connect
D/BluetoothSerialService(24567): connect to: 00:06:66:69:02:B9
D/BluetoothSerialService(24567): setState() 0 -> 2
I/BluetoothSerial(24567): MESSAGE_STATE_CHANGE: 2
I/BluetoothSerial(24567): BluetoothSerialService.STATE_CONNECTING
I/BluetoothSerialService(24567): BEGIN mConnectThread SocketType:Secure
I/BluetoothSerialService(24567): Connecting to socket...
W/BluetoothAdapter(24567): getBluetoothService() called with no BluetoothManagerCallback
D/BTIF_SOCK( 1945): service_uuid: 00001101-0000-1000-8000-00805f9b34fb
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
I/art     (  559): Explicit concurrent mark sweep GC freed 19649(1002KB) AllocSpace objects, 1(16KB) LOS objects, 33% free, 27MB/40MB, paused 1.983ms total 170.883ms
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
I/ActivityManager(  559): Killing 23783:com.google.android.gms.ui/u0a23 (adj 15): empty #17
W/libprocessgroup(  559): failed to open /acct/uid_10023/pid_23783/cgroup.procs: No such file or directory
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
W/bt-sdp  ( 1945): SDP - Rcvd conn cnf with error: 0x4  CID 0x47
E/bt-btif ( 1945): DISCOVERY_COMP_EVT slot id:8, failed to find channle,                                       status:1, scn:0
W/bt-btif ( 1945): invalid rfc slot id: 8
E/BluetoothSerialService(24567): java.io.IOException: read failed, socket might closed or timeout, read ret: -1
I/BluetoothSerialService(24567): Trying fallback...
W/BluetoothAdapter(24567): getBluetoothService() called with no BluetoothManagerCallback
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
V/ConfigFetchTask( 1316): ConfigFetchTask getDeviceDataVersionInfo(): ABFEt1Xo6zqLtIVgIpI7QimtbUwDknthH-LplxgWY5MC14aZHObgyS8ea7yes2qZgGfP2I_c7bIp8BSlCa-wPuxwWsGR9XGy2U4U95-UZQpHK5TiA7AsBWylmj6HLqpHAZY3EtVjvdr2_ceNEH9gDdPg6RN9UCaWVvzO6fuy0vN3BqJuzv3oalAvOaOr-B7FNBIlY2nRZL5r
I/GoogleHttpClient( 1316): Falling back to old SSLCertificateSocketFactory
D/HeadsetStateMachine( 1945): Disconnected process message: 10, size: 0
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
W/ActivityManager(  559): getTasks: caller 10138 does not hold GET_TASKS; limiting output
W/bt-rfcomm( 1945): PORT_StartCnf failed result:4
E/BluetoothSerialService(24567): Couldn't establish a Bluetooth connection.

rjwhite, 2015-01-24:
hmmm....so I wanted to conclusively prove that this was a problem with Android 5.0.1, so I installed my App on an old farty clunky Samsung GT-P7500R running 4.0.3, which your notes say your stuff works on (or a close enough model/version).  But, I got the same issue of not being able to connect to Bluetooth (I used connectInsecure() this time based on what you said/did in another Issue posted 'Pairing problem with Android #8'.  I did try connect() as well and did *not* get asked for the PIN, but I never seen this happen with 4.4.4 either).  This would make me suspect my Adafruit EZ-Link shield is now busted, but that's an amazing coincidence that it fried the same evening as my Android upgrade.  I can also still pair to the device, see it, etc.  So, I'm suspicious of my EZ-Link shield right now, but still not totally convinced.
Sorry if I've wasted your time.  This is extremely important to me because I have to demo this App real soon :-(

don, 2015-01-24:
I'd use connect() rather than connectInsecure() where possible. The Adafruit Bluefruit EZ-Link doesn't ask for a password. I'd expect the bluetooth radio is just in a bad state rather than broken, you should be able to get it working again.

rjwhite, 2015-01-25:
d'oh!!  Problem solved.  This is really embarrassing - it was a coding error in pattern matching and combination of chance events where I was trying to connect to the wrong Bluetooth MAC address.
Really poor timing that it manifested itself for the first time just as I upgraded to Android 5.0.1.
This issue can be closed.
I'm really sorry I wasted your time.  Thanx for all your feedback.  Sorry again!

don, 2015-01-25:
I'm glad you got it working!

closed 2015-01-25 by don