don/BluetoothSerial #103: Connecting to bluetooth devices

tridentNestea, 2015-01-27:
I'm having issues connecting to bluetooth devices through this plugin. When I try to connect to my Mac Mini I can see that a connection has been established through the bluetooth icon on the system tray. But after 15-20 seconds I get an error on the app saying that the connection was lost and on the mac the connection in the systems tray is also lost.

don, 2015-01-27:
I don't think this will work. The Android code expects the other device to provide Serial Port Profile (SPP) and have the RFCOMM UUID 00001101-0000-1000-8000-00805F9B34FB. I don't think your Mac Mini provides SPP support.

tridentNestea, 2015-01-28:
Okay thanks for the quick reply, I'll have to try find a different device for testing.

closed 2015-01-28 by tridentNestea