don/BluetoothSerial #104: Sending hex commands to an Arduino

testdevicetest, 2015-01-30:
Hi, I need to send hex commands to my Arduino Bluetooth device. Ie B4F6C6D1F2A8. I have the ability to connect to my device and send messages, but I'm not sure how to send commands like that. Any help would be appreciated, thanks for the cool plugin.

don, 2015-01-30:
I'd send that data as an Uint8Array.

	var data = new Uint8Array(6);
	data[0] = 0xb4;
	data[1] = 0xf6;
	data[2] = 0xc6;
	data[3] = 0xd1;
	data[4] = 0xf2;
	data[5] = 0xa8;	
	bluetoothSerial.write(data, success, failure);

See https://github.com/don/BluetoothSerial#write

testdevicetest, 2015-01-31:
Hey don, thanks for the help, I missed that in the documentation somehow. I haven't yet been able to get the response from the device that I was expecting using the above code. My success function is called so obviously the message is being sent. 

I know the device works so I guess I'm not sending the command in the proper format somehow. I'm new to this so I guess I'll just have to keep plugging away at it. I really appreciate the support you provide here, I'll leave this open for now in case you have any further advice.

closed 2015-02-02 by testdevicetest

umair975, 2016-11-28:
Hi,
I am trying to send ESC/POS commands to a bluetooth printer... I can send the HEX commands like u have mentioned above but how can I add text to it as well ... Ideally I would like to change the printer code page and then send arabic text string.