don/BluetoothSerial #106: Pairing?

stakutis, 2015-02-20:
Is there a way to pair a device (w/o using the OS's pairing utility)?

We have an app here that for some reason has it's own pairing screen to simplify the tasks for the user.  I'd rather not keep that special native code if I can find a solution in use by more than just my team...

don, 2015-02-20:
I've only used the OS pairing screens. I'm going to add an Android method for choosing unpaired devices, but the actually pairing would still happen with Android. For iOS and WP8 I think you're going to be stuck with whatever the OS gives you. I'd be interested to see how that app you have works.

stakutis, 2015-02-24:
Couple of quick questions:
- I see the "isEnabled()" function, but is there one to actually turn
bluetooth on (enable) or off?
- The software I'm trying to replace (all written in pure java and overly
ambitious whereas my new solution using your stuff is all javascript) has a
"discovery" function. It is slow.  How does you "list" function work?  It
is very fast. Does it go out and look or does it just ask the OS what was
last known?

Thanx.  So far, it is working great and I've ported a few medical devices
(which are notoriously quirky in their messaging noless).

Chris Stakutis
978 764 3488
Twitter: @Stakutis

don, 2015-02-24:
@stakutis there is no enable yet. I'm adding this functionality very soon. 

On Android `list` just asks the OS about paired devices. I still need to merge #100 (and rename the function) to add support for discovering unpaired devices on Android.

don, 2015-03-02:
[enable](https://github.com/don/BluetoothSerial#enable) and [discoverUnpaired](https://github.com/don/BluetoothSerial#discoverunpaired) have been added.

closed 2015-03-02 by don

stakutis, 2015-03-02:
Cool.  WIll it allow to DO a pairing?   I think i mentioned, i've
re-written a bunch of drivers that we had in all java for various medical
devices.  My new version(s) are about more than TWICE as fast with your
stuff (not sure why, and it is very noticable) and about 1/3 the code (plus
all jscript).

BUT, they have a need to do a pairing without the OS (the devices are
locked down).

Chris Stakutis
978 764 3488
Twitter: @Stakutis

stakutis, 2015-03-02:
Cool. I'm about to port to ios and later Windows mobile. Will your driver
work on both and appear the same to me? In the past I've written objective
c but hated it.

Chris Stakutis
978-764-3488

stakutis, 2015-03-02:
Also..If my medical devices are traditional Bluetooth will they not work
with ios or can ios handle both?

Chris Stakutis
978-764-3488

don, 2015-03-02:
Android will automatically begin the pairing process if you connect to an unpaired MAC address. (I think Windows Phone does the same thing.) Most of the API works the same across platforms. There are some differences. `enable` and `discoverUnpaired` only work on Android. `showBluetoothSettings` works on Android and Windows Phone.

You can't do Bluetooth Classic on iOS unless you join the [MFi program](https://developer.apple.com/programs/mfi/).

stakutis, 2015-03-02:
The driver has been working great for me and I've done a lot of testing
with different devices. My company may proceed in this path if we have
other clues about how it is working in the field. Any clue? And will the
license allow us?

Chris Stakutis
978-764-3488

don, 2015-03-03:
I'm glad it's working well for you. There are other people using this plugin in production.

The plugin is released under the [Apache License, version 2](http://www.apache.org/licenses/LICENSE-2.0)

stakutis, 2015-03-05:
The current internal/java code that we use (which I desperately want to
replace with this tool) will forcibly do a PAIRING if necessary (and for
each type of device we know the PIN or can fetch it from somewhere).  I
would *really* like to extend your Connect() call to also/optionally force
pairing.  I just dont know enough about the java/bluetooth stuff to do it.

In our code, we ultimately do something like this:

else if(BluetoothDevice.ACTION_PAIRING_REQUEST.equals(action)) {
   Log.i("ACTION_PAIRING_REQUEST","Pairing request");
   BluetoothDevice device =
intent.getParcelableExtra(BluetoothDevice.EXTRA_DEVICE);
   byte[] pinBytes = getPin();
   device.setPin(pinBytes);
    device.setPairingConfirmation(true);
}

But I'm sure there is more to it.  Any chance you can consider adding this
feature?

Chris Stakutis
978 764 3488
Twitter: @Stakutis