don/BluetoothSerial #107: subscribeRawData

zdravko, 2015-02-23:
I am writing an application for reading Nonin Oxymeter data.
It's binary data, not easy to handle.
Using Evothings Studio I built a project for Android. Many things work well, but:
bluetoothSerial.subscribeRawData(function (data) {...
returns:
TypeError: Object #<Object> has no method 'subscribeRawData'
The same happens on subscribeRaw only (no Data part).
This should be working.

don, 2015-02-23:
Are you seeing this error with the iOS or Android implementation?

zdravko, 2015-02-23:
Android.
I cant use iOS due to cordova error. I have win64 but it thinks it's 32 bit and cordova says that it can't set up project for iOS. I am not sure, I must have installed something wrong.

don, 2015-02-24:
I'm receiving data in my callback for bluetoothSerial.subscribeRawData. It looks like the first character is begin dropped so I need to debug more to see if the problem is in the plugin or my example.

Since you're getting a TypeError, I expect the plugin isn't installed correctly in your project

From the command line, verify that the plugin is installed

    C:\Users\Don\BluetoothSerialApiDev>cordova plugin ls
    com.megster.cordova.bluetoothserial 0.3.5 "Bluetooth Serial"

If the plugin is not listed, re-install it with 

    cordova plugin add com.megster.cordova.bluetoothserial

You'll need a Mac for iOS projects.

zdravko, 2015-02-24:
I'm confused. I can find subscribeRaw in the sources, but not
subscribeRawData as in Readme.
However, neither works. I'll try to figure out what is wrong.
Very strange, though. I added plugin by github URL instead of
by name com.megster....

zdravko, 2015-02-24:
This is frustrating. I can call any other function except 
subscribeRawData, unsubscribeRawData.
That is from my Evothings Studio JavaScript Workbench.
Since this is the only way to receive binary data I'm stuck.
To make things look even more strange, the output from this loop:

for (var key in bluetoothSerial) {
     hyper.log(key);
}
is:
LOG: connect
LOG: connectInsecure
LOG: disconnect
LOG: list
LOG: isEnabled
LOG: isConnected
LOG: available
LOG: read
LOG: readUntil
LOG: write
LOG: subscribe
LOG: unsubscribe
LOG: clear
LOG: readRSSI

No subscribeRawData!
Is there any other way to retreive array of integers from my device?

Thank you for the help.
Regards, Zdravko

Dne 24.2.2015 ob 4:04 je Don Coleman zapisal(a):

don, 2015-02-24:
@zdravko subscribeRawData is defined in bluetoothSerial.js. Based on what you posted it looks like you're running an old version of the plugin that doesn't include subscribeRawData.

Check which version you're running with the command line

    cordova plugin ls

If you uninstall and reinstall the plugin with cordova, you should get the latests.

    cordova plugin rm com.megster.cordova.bluetoothSerial
    cordova plugin add com.megster.cordova.bluetoothSerial

I'm not familiar with Evothings Studio JavaScript Workbench. Did it come with the plugin pre-installed?

zdravko, 2015-02-24:
It seems like Evothing issue. It has BLE pre-installed which also 
defines bluetoothSerial.
I will ask on their forum.
Z.

Dne 24.2.2015 ob 16:56 je Don Coleman zapisal(a):

don, 2015-02-24:
That makes sense. They have an older version included. Let me know what you find out and if there is any way to update.

don, 2015-03-02:
Not a bug. Evothings Studio uses an older version of this plugin without `subscribeRawData`

closed 2015-03-02 by don