don/BluetoothSerial #108: Error when adding plugin to VS

Gil401, 2015-02-23:
Hi @don 

When adding the plugin in VS 2015 it hangs - after restarting visual studio the plugin library is added to project library but the config.xml reference needs to be added manually.
<vs:plugin name="com.megster.cordova.bluetoothserial" version="0.3.5" />
after I am adding it, I get the following error:
1>MDAVSCLI : error : Uh oh!

this is the whole output:

1>------ Build started: Project: GymBETA, Configuration: Debug Android ------
1>C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v14.0\TypeScript\Microsoft.TypeScript.targets(96,5): warning : The TypeScript Compiler was given no files for compilation, so it will skip compiling.
1>  GeneratedJavascript=
1>  C:\Users\Gil\Source\Repos\G8B\GymBETA>call "C:\Program Files (x86)\nodejs\"\nodevars.bat 
1>  Your environment has been set up for using Node.js 0.10.33 (ia32) and npm.
1>  ------ Ensuring correct global installation of package from source package directory: C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\Extensions\ymbsackx.r2q\packages\vs-mda
1>MDAVSCLI : error : Uh oh!
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

don, 2015-02-24:
@Gil401 Can you point me to instructions for getting Cordova support added to Visual Studio?

Gil401, 2015-02-24:
Hi @don 
You can just download it from here:
http://www.visualstudio.com/en-us/downloads/visual-studio-2015-ctp-vs

don, 2015-02-24:
Thanks. I didn't realize cordova support was built in. I'll take a look.

Gil401, 2015-02-26:
Hi @don ,

Any findings? :)

don, 2015-02-28:
@Gil401 This worked OK for me. 

I installed VS 2015 CTP6 on Windows 10 and updated all the Cordova tools.

I created a new Blank Cordova Project using the VS wizard.

I used the VS GUI for config.xml and added a custom plugin from a git URL https://github.com/don/BluetoothSerial

I edited `scripts\index.js` and added the following inside `function onDeviceReady`

        var success = function (devices) {
            alert(JSON.stringify(devices));
        }

        var failure = function (reason) {
            alert("Listing Bluetooth Devices Failed " + reason);
        }

        bluetoothSerial.list(success, failure);

The code deployed and ran on my Windows Phone 8.1. It wouldn't deploy to my Nexus 5 because I need to install a driver for it to recognize the phone.

Hopefully that helps you get things working.

closed 2015-03-12 by don

icaro56, 2015-07-06:
To solve this you have that to remove the platform android and the file android.json inside plugins. Generate again the build and everething works.