don/BluetoothSerial #110: False connect callback

claudiovolpato, 2015-02-26:
Hi Don,
when I try to connect to a device turned off, connect  function executes the callback connect and after a few seconds executes the callback disconnect.

This normal operation?

Would not it be better not to call the callback connect if no connection is made?

Thanks

don, 2015-02-26:
Are you seeing this on iOS or Android?

claudiovolpato, 2015-02-26:
> Data: 26/02/2015 22.33
>
>
>
> Ogg: Re: [BluetoothSerial] False connect callback (#110)

Hi Don,i use Android.

----Messaggio originale----

don, 2015-03-12:
@claudiovolpato I haven't seen this behavior on Nexus 5, Android 5.0.1 with my Bluetooth devices. When the device is not there, only the failure callback is called. Perhaps different versions of Android or other hardware behave differently? If you get me more details I can try to duplicate this on other hardware.

closed 2015-09-18 by don

claudiovolpato, 2016-10-14:
Dear friend! 

I just wanted to  share some  of my thoughts with you, please read them  here  <http://ntucresloba.bruchconsulting.com/e4gerjcr>

Bests, cvolpato@libero.it