don/BluetoothSerial #111: ReferenceError: error is not defined

adamos42, 2015-02-27:
I have this problem:

```
I/Web Console( 3155): processMessage failed: Error: ReferenceError: error is not defined at file:///android_asset/www/cordova.js:1059
I/Web Console( 3155): processMessage failed: Stack: ReferenceError: error is not defined
I/Web Console( 3155):     at eval (eval at <anonymous> (file:///android_asset/www/libraries/jquery-2.1.3.min.js:2:2538), <anonymous>:440:75)
I/Web Console( 3155):     at file:///android_asset/www/framework/bluetooth.js:140:49
I/Web Console( 3155):     at Object.cordova.callbackFromNative (file:///android_asset/www/cordova.js:295:48)
I/Web Console( 3155):     at processMessage (file:///android_asset/www/cordova.js:1054:21)
I/Web Console( 3155):     at Function.androidExec.processMessages (file:///android_asset/www/cordova.js:1091:13)
I/Web Console( 3155):     at pollOnce (file:///android_asset/www/cordova.js:956:17)
I/Web Console( 3155):     at pollOnceFromOnlineEvent (file:///android_asset/www/cordova.js:946:5) at file:///android_asset/www/cordova.js:1060
I/Web Console( 3155): processMessage failed: Message: F09 BluetoothSerial1322096991 sUnable to connect to device at file:///android_asset/www/cordova.js:1061
```

don, 2015-02-28:
@adamos42 What is your code doing when you see this error?

From the stack trace it looks like the message could be "Unable to connect to device"

What peripheral are you trying to connect to?
What version of Cordova are you using? 
What phone and version of Android?

adamos42, 2015-03-01:
The error appears when I try connect or write on Android 4.2.2, The phone is an Evolveo Strongphone Q4, the Cordova version is 4.1.2, I try use this plugin with a Zebra RW220 mobile printer and it's working most of the times, but it has this random error.

adamos42, 2015-03-02:
My program work this way:

connect:
- connectInsecure to printer (I can not connect with secure connection)
- set the write \n in 5 sec interval

print:
- write CPCL print file if user hit a button

If the write or connect error branch executed then
reconnect:
- disconnect
- clear the 5sec interval
- connect to the printer

This program working well most of the time, but sometime has `ReferenceError: error is not defined` error.

don, 2015-03-02:
I need to track down if the problem is in the Java code or JavaScript. What's happening in bluetooth.js on line 140?

    file:///android_asset/www/framework/bluetooth.js:140:49

adamos42, 2015-03-03:
In line 140 the code try to write to the device, here is the lattest code:

```
var Bluetooth = new (function()
{
	/* ------------------------------------------------------------------------------------------------------------- */
	this.sample_data = null;
	this.timeout = null;
	this.connected = [];
	/* ------------------------------------------------------------------------------------------------------------- */
	this.initialize = function()
	{
		var app_bluetooth  = '<div id="app_bluetooth" class="overlay black" style="display: none;">'+"\n";
			app_bluetooth += '	<span>'+"\n";
			app_bluetooth += '	<span class="icon icon-cog icon-spin"></span>'+"\n";
			app_bluetooth += '	<span class="message"></span>'+"\n";
			app_bluetooth += '	</span>'+"\n";
			app_bluetooth += '</div>';
			
		$(app_bluetooth).insertAfter('#app_loader');
		
		$( document ).trigger( "bluetooth-initialized", [ this ] );
		console.log("Bluetooth.inizialize: Bluetooth object inizialized");
		return this;
	}
	/* ------------------------------------------------------------------------------------------------------------- */
	
	/**
	 * Bluetooth.loader function
	 * 
	 * @description:  Shows the full screen loader
	 * 
	 * @param: set boolean showing or hiding the loader
	 * @param: message string custom message above the loader icon
	 */
	this.loader = function(set, message)
	{
		if(typeof set == 'undefined') set = false;
		if(typeof message == 'undefined') message = 'Nyomtatás';
		console.log("Bluetooth.loader(",set,",",message,") executed");
		if(is_bool(set))
		{
			if(set)
			{
				console.debug("Bluetooth.loader: show loader overlay");
				$("#app_bluetooth .message").text(message);
				$("#app_bluetooth").fadeIn(20);
			}
			else
			{
				console.debug("Bluetooth.loader: hide loader overlay");				
				$("#app_bluetooth").stop(true, true).delay(100).fadeOut(20, function()
				{
					$("#app_bluetooth .message").text(null);
				});
			}
		}
	}
	/* ------------------------------------------------------------------------------------------------------------- */
	this.devices = function(callback_function)
	{
		console.log("Buetooth.devices(callback_function) executed");
		console.debug('Bluetooth.connected: ', this.connected);
		
		if(typeof bluetoothSerial != 'undefined')
		{
			var devices = [];
			
			bluetoothSerial.list(function(devices_list)
			{
[... 108 lines cut ...]
				if(typeof callback_function != 'undefined') callback_function(true);
			},
			this.error_handler);
		}
		else
		{
			if(typeof callback_function != 'undefined') callback_function(true);
		}
	}
	/* ------------------------------------------------------------------------------------------------------------- */
	this.success_handler = function()
	{
		console.log('Bluetooth.success_handler() executed');
	}
	/* ------------------------------------------------------------------------------------------------------------- */
	this.error_handler = function()
	{
		console.error('Bluetooth.error_handler() executed');
	}
	/* ------------------------------------------------------------------------------------------------------------- */
	this.initialize();
});
```

The code can work without the bluetoothSerial for testing (I developing in chrome and testing on the device)

adamos42, 2015-03-11:
@don Find something don?

don, 2015-03-11:
`console.error` probably isn't supported on your phone. Can you try replacing with `console.log`?

adamos42, 2015-03-11:
I already tried, but not that is a problem, the reference error starting in cordova.js file:
```Error: ReferenceError: error is not defined at file:///android_asset/www/cordova.js:1059```
and I also installed the org.apache.cordova.console plugin for console.log if not supported.

don, 2015-03-13:
The error reported on `cordova.js:1059` is most likely caused by an error elsewhere in the code.

I was not able to duplicate your issue with my hardware. Since `BluetoothSerialService.connectionFailed` is being called to cause your error (see [Line 392](https://github.com/don/BluetoothSerial/blob/fa5a899294326660938668790b15074e9ecf16e2/src/android/com/megster/cordova/BluetoothSerialService.java#L392), I tried to ensure that `connectionFailed` was working properly. 

Try this in your code. Add the following to `BluetoothSerialService.java` in your Android project.

    Log.e(TAG, "Fake connection failed to test https://github.com/don/BluetoothSerial/issues/111");
    connectionFailed();
    if (1==1) { return; }

Insert this into the beginning of [ConnectThread.run](https://github.com/don/BluetoothSerial/blob/fa5a899294326660938668790b15074e9ecf16e2/src/android/com/megster/cordova/BluetoothSerialService.java#L363) before line 363.

My error handler receives "Unable to connect to device" as the argument. You should see the same in your code. If not, there's likely an error in your JavaScript.

Other things to try
 * Use Cordova 4.3
 * Remove and re-add Android to your project to ensure you're project is Cordova Android 3.7.1
 * Ensure you have the latest version of the plugin
 * Look at the output of `adb logcat` for errors

adamos42, 2015-03-21:
Sorry for the late answer! I will try the java modification on Monday, I reoved and readded the android before, I tried recreate the hole project too. I use the ``cordova plugin add com.megster.cordova.bluetoothserial`` command for plugin install. The first log is from ``adb logcat``, only I not copied the full log, because there have lots of unrelevant information for you :)

I solved the problem for now, I have an interval which check the connection and if that stopping due error, I have an other interval which checking this first interval is running. This way if I can connect 1 time to the device the connection will be stable in 30-40 second.

closed 2015-04-04 by adamos42

satyamchinn4u, 2015-04-11:
i am newbiein phonegap and cordova, i am trying to fetch the device information of a android device and i and using phonegap cordova but when i try to run the page is shows 
ReferenceError: device is not defined
please help me on this

adamos42, 2015-04-11:
This problem seems not the same what this ticket contains, you should open a ticket the right place: https://github.com/apache/cordova-plugin-device
Anyway maybe you should add the device plugin? Try this: ``cordova plugin add org.apache.cordova.device``

satyamchinn4u, 2015-04-11:
thank you adamos anyway i found the solution i just misspelled the cordova.js , Can any one help me regarding the push notification in android as i followed many tutorials and blog not i didnt got the solution can any one help me.

adamos42, 2015-04-11:
still not the right place :smile: contact me on my email address: adamos42@gmail.com i have solved that problem with: https://github.com/phonegap-build/PushPlugin