don/BluetoothSerial #112: Pair (bind) and UnPair with PIN too!

stakutis, 2015-03-11:
Hi! I've updated the .java code and added two new callable exec functions (pair, unpair) which now allows the app to programatically pair a device even with a PIN code ... so, no user-interaction which is important for our application.

How can I get this put into the main code stream?  Do I send it to you or ...?

don, 2015-03-11:
Fork this project. Create your changes on a branch in your repo, push to github and issue a pull request.

https://guides.github.com/activities/forking/

stakutis, 2015-03-17:
Windows-WP8...quick question.  I'm just starting to play with it and I
can't get it take any commands.  Says something about "must use Start
first".  I think it is coming from the lower blue stuff.  Is there a config
file like on Android where you tell it what you need for privs?

Chris Stakutis
978 764 3488
Twitter: @Stakutis

don, 2015-04-08:
@stakutis can you check the comments in the pull request #114?

jitendracodehacker, 2019-08-23:
any update on this to accommodate in the master branch as well as in ionic-native packages ?

jitendracodehacker, 2019-08-27:
@don  & @stakutis  : could you guide me how to fix the issue of ionic-native api error