don/BluetoothSerial #113: bluetoothSerial.connect() causes the app to freeze

jep88, 2015-03-11:
Hi Don!

I've been trying to get this to work and have my android phone (version 4.2.2) communicate with a bluetooth device, and I am experiencing some issues.

The device I am using is an Engduino with a bluetooth module. This is our university's modification of the Arduino and the HC-05 bluetooth module, and can be treated as such.

I have been going through the examples, but couldn't get any of them to work, and the issue is that whenever I call bluetoothSerial.connect() the app basically freezes - the connection neither succeeds nor fails, otherwise it would have called the appropriate functions (which it doesn't, it just sits there). You can push buttons and things like that in this state, but they don't do anything.

What is interesting is that if I supply the wrong mac address, the connection will fail and the appropriate function is called displaying a message that the connection has failed. The behavior above is only observed when I call bluetoothSerial.connect() with the correct mac address.

Just to note, I have tried using other apps to connect to my hardware and they have all worked flawlessly (BlueArduino and Bluetooth Terminal on android and BT Terminal on windows phone), and I have flashed both the sketches in your examples and other code for bluetooth communication onto my device. This leads me to think that the issue is unlikely to be on the hardware side of things.

Keep in mind that I am very much new to all of this (cordova, bluetooth, arduinos), so if my explanation wasn't clear, please tell me and I'll try to clarify what I can.

don, 2015-03-11:
Your explanation makes sense. I'd guess there's an unexpected condition in the Java code so the success callback (or error callback) is not being called.

Can you run `adb logcat` while connecting and see if there's a stack trace with more information?

jep88, 2015-03-12:
Thank you for the quick response!

I used logcat with the Counter example app, my only modification being changing the mac address value to the appropriate one.

I personally had trouble getting any info from the log. All I could gather is that it gets stuck when it calls CordovaActivity.onMessage(). I looked at the source code for that method and that didn't help much either. So I am giving the logs to you, perhaps you can get something out of them.

I filtered the output to only show stuff from the "com.megster.bluetooth.count" application, and made two versions - one filtered as "debug" and one verbose (I don't actually know what "debug" is supposed to filter out, but just in case I also included the verbose version, which is about 40 lines longer).

The log starts when I start up the application, and ends when it gets stuck trying to connect (logcat actually stopped outputting anything once it reached what is at the end of the log, unless I did something like close the app).

And because among other things I am also new to github, so I don't know how to upload those logs to you using github in some way, so I just uploaded them to google drive and you can get them here:

https://drive.google.com/file/d/0B7rha48ADySsV1ZIdF92WEpzdnM/view?usp=sharing

Looking forward to your response!

don, 2015-03-12:
It looks like the plugin isn't installed.

    03-12 01:21:51.977: D/dalvikvm(22051): NOTE: loadClass 'com.megster.cordova.BluetoothSerial' 0x41f8c000 threw exception Ljava/lang/ClassNotFoundException;
    03-12 01:21:51.978: W/System.err(22051): java.lang.ClassNotFoundException: com.megster.cordova.BluetoothSerial
    03-12 01:21:51.979: W/System.err(22051): 	at java.lang.Class.classForName(Native Method)
    03-12 01:21:51.980: W/System.err(22051): 	at java.lang.Class.forName(Class.java:217)
    03-12 01:21:51.980: W/System.err(22051): 	at java.lang.Class.forName(Class.java:172)
    03-12 01:21:51.980: W/System.err(22051): 	at org.apache.cordova.PluginManager.instantiatePlugin(PluginManager.java:403)
    03-12 01:21:51.980: W/System.err(22051): 	at org.apache.cordova.PluginManager.getPlugin(PluginManager.java:193)
    03-12 01:21:51.980: W/System.err(22051): 	at org.apache.cordova.PluginManager.exec(PluginManager.java:141)
    03-12 01:21:51.980: W/System.err(22051): 	at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:63)
    03-12 01:21:51.980: W/System.err(22051): 	at org.apache.cordova.ExposedJsApi.exec(ExposedJsApi.java:40)
    03-12 01:21:51.981: W/System.err(22051): 	at android.webkit.JWebCoreJavaBridge.sharedTimerFired(Native Method)
    03-12 01:21:51.981: W/System.err(22051): 	at android.webkit.JWebCoreJavaBridge.sharedTimerFired(Native Method)
    03-12 01:21:51.981: W/System.err(22051): 	at android.webkit.JWebCoreJavaBridge.fireSharedTimer(JWebCoreJavaBridge.java:92)
    03-12 01:21:51.981: W/System.err(22051): 	at android.webkit.JWebCoreJavaBridge.handleMessage(JWebCoreJavaBridge.java:108)
    03-12 01:21:51.981: W/System.err(22051): 	at android.os.Handler.dispatchMessage(Handler.java:99)
    03-12 01:21:51.981: W/System.err(22051): 	at android.os.Looper.loop(Looper.java:153)
    03-12 01:21:51.981: W/System.err(22051): 	at android.webkit.WebViewCore$WebCoreThread.run(WebViewCore.java:919)
    03-12 01:21:51.982: W/System.err(22051): 	at java.lang.Thread.run(Thread.java:838)

Can you try

    cd Counter
    cordova plugin ls
    cordova plugin add com.megster.cordova.bluetoothserial
    cordova run android --device

jep88, 2015-03-12:
I tried what you said and it still doesn't work.

Doing "cordova plugin ls" gives me:
com.megster.cordova.bluetoothserial 0.4.1 "Bluetooth Serial"

So it probably has the plugin installed. I still tried adding the plugin, and as expected it told me:
Plugin "com.megster.cordova.bluetoothserial" already installed on android.

I then did "cordova run android --device" and ran it on my phone, but it still behaves in the exact same way, and it still gives this in the log:

03-12 16:42:03.493: W/System.err(8670): java.lang.ClassNotFoundException: com.megster.cordova.BluetoothSerial

Perhaps there is an issue with how I'm building it? Initially I used eclipse and after you suggested using "cordova run" I hoped that maybe cordova will know how to build the app properly, but apparently not.

Here is the full log for this attempt:
https://drive.google.com/file/d/0B7rha48ADySsbXRTd2o2a0RrTEE/view?usp=sharing

don, 2015-03-12:
It looks like you're having the same problem. Cordova can't find the plugin. Something must be installed incorrectly.

    03-12 16:42:03.492: D/dalvikvm(8670): NOTE: loadClass 'com.megster.cordova.BluetoothSerial' 0x41f8bfc8 threw exception Ljava/lang/ClassNotFoundException;
    03-12 16:42:03.493: W/System.err(8670): java.lang.ClassNotFoundException: com.megster.cordova.BluetoothSerial
    03-12 16:42:03.493: W/System.err(8670): 	at java.lang.Class.classForName(Native Method)
    03-12 16:42:03.493: W/System.err(8670): 	at java.lang.Class.forName(Class.java:217)
    03-12 16:42:03.494: W/System.err(8670): 	at java.lang.Class.forName(Class.java:172)
    03-12 16:42:03.494: W/System.err(8670): 	at org.apache.cordova.PluginManager.instantiatePlugin(PluginManager.java:403)

Let's try an eliminate Eclipse from the picture for now. You can always go back to Eclipse once everything is working. 

Can you try with a clean copy of the code?

    git clone https://github.com/don/BluetoothSerial.git
    cd BluetoothSerial/examples/Counter
    cordova platform add android
    cordova plugin add com.megster.cordova.bluetoothserial
    cordova run android --device

jep88, 2015-03-12:
It worked! Thank you Don!

But I would also like to know exactly what was it that I was doing wrong. Is the order in which I execute the commands above actually significant? Because sometimes I would add the plugin before adding the android platform, and sometimes I would first edit the mac address in the javascript file before adding the android platform.

Also I always took the examples outside of the BluetoothSerial folder, because I assumed that adding the plugin should add all the necessary files, perhaps this is not the case?

But thanks again for the help!

don, 2015-03-12:
Plugins and platforms can be added and removed in any order.

Moving the examples out of the BluetoothSerial folder is a good idea. The example can run inside the BluetoothSerial folder **if** you install the plugin from http://plugins.cordova.io with `cordova plugin install com.megster.cordova.bluetoothserial`. If you try and install a local version of the plugin, it will fail when the example is nested under the plugin. e.g. `cordova plugin install ~/BluetoothSerial`.

I don't know what happened to your original project. Perhaps uninstalling and reinstalling the plugin would help.

    cordova plugin rm com.megster.cordova.bluetoothserial
    cordova plugin add com.megster.cordova.bluetoothserial

 Removing and re-installing the platform can also help. It's possible Eclipse just broke something

    cordova platform rm android
    cordova platform add android

closed 2015-03-12 by don