don/BluetoothSerial #115: Add a 'browser' target for emulated bluetooth

PhracturedBlue, 2015-03-13:
I've been trying to write a serialBluetooth application, and find development to be much faster when using the browser target.  To allow me to develop the bluetooth API for my app, I decided to enhance this plugin with a 'browser' target.

Note that not all bluetoothSerial functions are yet supported, but read/write functions shoul work properly

The code is all asynchronous to try to simulate a real bluetooth device as closely as possible
When using the browser, it is necessary to have a processing function that can eulate the bluetooth device.  To do so, I added a new 'register' function as follows:
```
bluetoothSerial.register(function(buf) {
     //buf.input is the data that was received via bluetoothSerial.write
     //buf.output is data that will be transmitted via a bluetotthSerial.read or subscribe
     //Do processing here
     buf.input = ""
});
```

Here is the actual patch:
```
diff --git a/plugin.xml b/plugin.xml
index 8f8a532..856f5e0 100644
--- a/plugin.xml
+++ b/plugin.xml
@@ -83,4 +83,12 @@
          <source-file src="src/wp/BluetoothConnectionManager.cs" />
 
     </platform>
+
+    <!-- browser -->
+    <platform name="browser">
+        <js-module src="www/browser/bluetoothSerial.js" name="BluetoothSerial_browser">
+            <clobbers target="window.bluetoothSerial" />
+        </js-module>
+
+    </platform>
 </plugin>
diff --git a/www/browser/bluetoothSerial.js b/www/browser/bluetoothSerial.js
new file mode 100644
index 0000000..60bc04b
--- /dev/null
+++ b/www/browser/bluetoothSerial.js
@@ -0,0 +1,150 @@
+window.bluetoothSerial = (function() {
+    var connected = false;
+    var enabled = true;
+    var buf = {
+        input: "",
+        output: "",
+        };
+    var subscribe_cb = function(value) {};
+    var raw_cb = false;
+    var subscribe_delim = false;
+    var interval;
+    var process_cb;
+
+    var btlog = function(str) { console.log(str); }
+    var timer_cb = function() {
+        if(process_cb) {
+            process_cb(buf);
+        }
+        if (buf.output.length) {
+            if(subscribe_delim) {
+                var index = buf.output.indexOf(subscribe_delim);
+                if (index > -1) {
+                    var data = buf.output.substr(0, index+subscribe_delim.length);
+                    buf.output = buf.output.substr(index+subscribe_delim.length);
+                    subscribe_cb(data);
+                }
+            }
+        }
+    };
+    return {
+        connect : function(mac, success_cb, fail_cb) {
+            btlog("bluetoothSerial.connect: " + mac);
+            connected = true;
+            if (success_cb) { success_cb(); }
+        },
+        register : function(data_cb) {
+            interval = window.setInterval(timer_cb, 100);
+            process_cb = data_cb;
+        },
+        disconnect : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.disconnect");
+            connected = false;
+            window.clearInterval(interval);
+            if (success_cb) { success_cb(); }
+        },
+        write : function(data, success_cb, fail_cb) {
+            btlog("bluetoothSerial.write: " + data);
+            buf.input += data;
+            if(success_cb) { success_cb(); }
+        },
+        available : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.available");
+            success_cb(buf.output.length);
+        },
+        read : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.read: " + buf.output);
+            var data = buf.output;
+            buf.output = "";
+            success_cb(data);
+        },
+        readUntil : function(delimiter, success_cb, fail_cb) {
+            btlog("bluetoothSerial.readUntil");
+            var index = buf.output.indexOf(delimiter);
+            if (index == -1) {
+                success_cb("");
+            } else {
+                var data = buf.output.substr(0, index+subscribe_delim.length);
+                buf.output = buf.output.substr(index+subscribe_delim.length);
+                success_cb(data);
+            }
+        },
+        subscribe : function(delimiter, success_cb, fail_cb) {
+            btlog("bluetoothSerial.subscribe '"+delimiter+"'");
+            subscribe_cb = success_cb;
+            subscribe_delim = delimiter;
+        },
+        unsubscribe : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.unsubscribe");
+            subscribe_delim = false;
+            if(success_cb) { success_cb(); }
+        },
+        subscribeRawData : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.subscribeRawData");
+            raw_cb = success_cb;
+        },
+        unsubscribeRawData : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.unsubscribeRawData");
+            raw_cb = false;
+            if(success_cb) { success_cb(); }
+        },
+        clear : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.clear");
+            buf.output = "";
+            if(success_cb) { success_cb(); }
+        },
+        list : function(success_cb, fail_cb) {
+            alert("bluetoothSerial.list is not implemented");
+        },
+        isConnected : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.isConnected: " + connected);
+            if(connected) {
+                if(success_cb) { success_cb(); }
+            } else {
+                if(fail_cb) { fail_cb(); }
+            }
+        },
+        isEnabled : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.isEnabled: " + enabled);
+            if(enabled) {
+                if(success_cb) { success_cb(); }
+            } else {
+                if(fail_cb) { fail_cb(); }
+            }
+        },
+        readRSSI : function(success_cb, fail_cb) {
+            alert("bluetoothSerial.readRSSI is not implemented");
+        },
+        showBluetoothSettings : function(success_cb, fail_cb) {
+            alert("bluetoothSerial.showBluetoothSettings is not implemented");
+        },
+        enable : function(success_cb, fail_cb) {
+            btlog("bluetoothSerial.enable");
+            enable = true;
+            if(success_cb) { success_cb(); }
+        },
+        discoverUnpaired : function(success_cb, fail_cb) {
+            alert("bluetoothSerial.discoverUnpaired is not implemented");
+        },
+    }
+})();
+module.exports.connect               = window.bluetoothSerial.connect;
+module.exports.register              = window.bluetoothSerial.register;
+module.exports.disconnect            = window.bluetoothSerial.disconnect;
+module.exports.write                 = window.bluetoothSerial.write;
+module.exports.available             = window.bluetoothSerial.available;
+module.exports.read                  = window.bluetoothSerial.read;
+module.exports.readUntil             = window.bluetoothSerial.readUntil;
+module.exports.subscribe             = window.bluetoothSerial.subscribe;
+module.exports.unsubscribe           = window.bluetoothSerial.unsubscribe;
+module.exports.subscribeRawData      = window.bluetoothSerial.subscribeRawData;
+module.exports.unsubscribeRawData    = window.bluetoothSerial.unsubscribeRawData;
+module.exports.clear                 = window.bluetoothSerial.clear;
+module.exports.list                  = window.bluetoothSerial.list;
+module.exports.isConnected           = window.bluetoothSerial.isConnected;
+module.exports.isEnabled             = window.bluetoothSerial.isEnabled;
+module.exports.readRSSI              = window.bluetoothSerial.readRSSI;
+module.exports.showBluetoothSettings = window.bluetoothSerial.showBluetoothSettings;
+module.exports.enable                = window.bluetoothSerial.enable;
+module.exports.discoverUnpaired      = window.bluetoothSerial.discoverUnpaired;
+
```

don, 2015-12-03:
@PhracturedBlue thanks for submitting this. It will be in the next release.

closed 2015-12-03 by don

luckebar, 2018-10-27:
how can I simulate a response from the device with this implementation? Could you post an example? Thank you