don/BluetoothSerial #116: Failure using on WP8

stakutis, 2015-03-18:
Hi...I'm trying to get my test-app to work on a new Windows 8 phone.  Some of the basic functions (like isEnabled()) work, but the main ones return this error:  FindAllPeerSync cannot be called before Start.

Anyone have any idea what that means?  My "capabilities" looks set correctly.  I'm not a WP8 expert so  I'm clueless...

don, 2015-03-18:
Try calling `list` before `connect`.

If that doesn't work, can you make you test code available?

stakutis, 2015-03-18:
Cool. List was failing and so I took off the Z in the function and it works
now. No idea why. I also tried calling Start() per some web examples and
that did not work and made things worse.

I think my phone is 8.1. Maybe that makes things different.

Connect still fails. Might parenthesis around mac help?

My code...It's nothing. I'm using console to just call your functions for
fun. I really want to get this to work. Microsoft is hugely keeen on our
business as we ship like thousands and thousands of devices a month. World
wide and wp is a lot more prevalent overseas than here.

Chris Stakutis
978-764-3488

don, 2015-03-18:
That 'Z' looks like a typo that got committed. I'll find a WP8 device and test tonight. The WP code is pretty new.

I think the parenthesis around the mac address are optional, but I'm not sure.

don, 2015-03-18:
I pushed version 0.4.2 removing the typo in the WP8 code.

You can try https://github.com/don/BluetoothSerialApiDev to exercise the API on WP8

stakutis, 2015-03-19:
(using my version...same at this point) I put parens around the mac and it
worked!!! I dumped the data from my device!!  No idea why, but i saw some
comment from you (I think) that parens might be a good thing.  Odd.

I'll do more tomorrow...right now I'm happy.

Chris Stakutis
978 764 3488
Twitter: @Stakutis

stakutis, 2015-03-19:
Things are working well now.  I have 2 different blue tooth devices passing
all my tests.  Things seem slower on Windows WP8 though, but i'm not really
sure.  On android, it connects faster and will re-connect faster (if it
loses connection, i auto-retry).  But it is working.  Very cool...because
we've written these pseudo-drivers for about a million different languages
to-date, and now, just javascript!!!!

Chris Stakutis
978 764 3488
Twitter: @Stakutis

stakutis, 2015-03-25:
I have a customer sending me their pedometer code that they claim is not
working well on 4.4.3 but fine on 5.0.   I know they wrote the java code
themselves.  I'd like to take a stab with the bluetooth driver..BUT...I'm
confused.  They say their device is BLE...Android supports that? How about
with this driver?  I'm confused...

Chris Stakutis
978 764 3488
Twitter: @Stakutis

don, 2015-03-25:
This plugins supports Bluetooth Classic on WP8 and Android and BLE on iOS.

For BLE on Android use [my BLE central](https://github.com/don/cordova-plugin-ble-central).

Android 4.3 and 4.4 support BLE but it's pretty buggy. 5.0 is better. Rebooting the device (or restarting Bluetooth and WiFi) often fixes many BLE issues on Android.

stakutis, 2015-03-27:
Thanx man. The manufacturer (from Barcelona) sent me their pedometer, the
BLE interface guide, and their sample java code.  I'm new to BLE but I want
to whip-up a javascript/phonegap test-environment, and quickly.

I have some really dumb questions (and I'm still amazed that my Android
will support BLE, but then again, I wear a fitbit and I'm pretty sure that
is BLE as well).  How do I "pair"?  I scan on my phone and it doesn't show
up...

Chris Stakutis
978 764 3488
Twitter: @Stakutis

don, 2015-03-27:
There is no pairing with BLE.

 * Scan for peripherals (ideally restricting to devices advertising services you care about)
 * Connect for a peripherals
 * Discover the device's characteristics and services
 * Subscribe to be notified when a characteristic value changes (optional)
 * Read and Write characteristic values (optional)

Some characteristics have permissions requiring elevated encryption. If necessary the peripheral and phone will negotiate an authenticated connection similar to bluetooth classic pairing.

closed 2015-03-27 by don

stakutis, 2015-03-28:
Thanx...that was helpful (because I'm still new to BLE syntax and style).
The "guide" I have for this device is not well written. I have the
customer's java code/implementation, which is somewhat helpful, but I'm
stuck.

Help...

I'm trying to do the most-basic things first, like, ask it for its software
version or battery life.  I find the service UUID.  Their document lists a
bunch of charactteristic UUIDs, and I see those reported, which is good.
But they dont give a lot of description on what each does.

They have a whole separate section that talks about sending commands to the
device. Easy-enough format...bunch of bytes...commands, byte-count,
chksum...child stuff (and I convert it, per your document).  But it is
unclear to me to "which" characteristic to send that to.  In their java
code, I see the one they use. So here's what I know:

- If I send a write command to a bad-spelled service or characterstic, it
fails, like you'd expect.
- Which sorta means if it doesn't fail, i've done those right.
- If i send "with not response" i get back the string "OK" on send...but
nothing more.
- If i send expecting a response, the "fail()" gets called with a value
'3'...no idea what that means.

I have also setup a "startNotification" on another characteristic that is
supposed to be the one to get status changes in the pedometer (like,
steps-taken-today), per their java code. But it never gets hit.

I'm a loss and really want to at least get this limping in javascript.  If
i can get it to work, I can help them with their real problem I'm sure.
But i've got to be able to talk to the device to start, right??  Uhg.  I'm
at a loss.

Chris Stakutis
978 764 3488
Twitter: @Stakutis

stakutis, 2015-03-30:
So I made progress on that pedometer on BLE on Android.  It was really a
pain to figure out their documentation.  And here's the problem...when you
ask it for all the step/data, it issues a bunch of notifications in a row
(supposedly I should accumulate then parse) but then it just stops.  Stops.
Goes away.  Doesn't finish with proper last packet, and you have to go back
into Connect mode after waking it up.

Is there some BLE timeout thing that might make it just shut down like
that?  There is another command, "Get Status", that I can call
over-and-over in a loop and seems to run forever.

Chris Stakutis
978 764 3488
Twitter: @Stakutis

don, 2015-04-02:
@stakutis this is getting pretty far away from the original issue, perhaps we should move this discussion to another forum.

It sounds like you have the mechanics of BLE correct, but you need to figure out what they have intended with your hardware. Ideally this can be done with documentation, but often it requires reverse engineering.

Use a tool like [LightBlue](https://itunes.apple.com/us/app/lightblue-bluetooth-low-energy/id557428110?mt=8) or [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to explore the characteristics and properties.

It's possible with the notifications stopping that the device is expecting you to write to another characteristic? Some BLE devices (incorrectly IMO) implement token based flow control.  It sounds like the BLE stack is hanging or crashing. (Likely on older Android versions.) Or, it could be the BLE device crashed. A BLE debugger may help.  I use the [Adafruit Bluefruit LE Sniffer](http://www.adafruit.com/product/2269) and Wireshark. Android also has "Enable Bluetooth HCI snoop log" in the debug settings, but I haven't tried that yet.