don/BluetoothSerial #117: can't find variable: bluetoothSerial

shadowmouse, 2015-03-24:
I've been banging my head against this issue for a few days now and figured I should ask around to see what I'm missing. Details as follows :

-- Setup -- 
Evothings Workbench 1.1.3
iOS 8.2
SublimeText 2

Whenever I try to call the list function I get the following error : 

"[ERR] ReferenceError: Can't find variable: bluetoothSerial"

I've done as much digging as I can and seem to be going in circles. The following code is where the issue occurs. As far as I've been able to determine all the other Cordova components and plugins are loading properly but the bluetoothSerial object is just missing flat from the window object.

<code>
document.addEventListener("deviceready", function () {

	bluetoothSerial.list(function (list) {
		hyper.log(JSON.stringify(list));
	}, function (err) {
		hyper.log(JSON.stringify(err));
	});

}, false);
</code>

I know the bluetooth is working because the evothings.ble module is listing the nRF8001 + arduino combo (Adafruit Breakout Board for the nRF8001) I'm using and I can properly connect to it. Unfortunately I can't get the evothings.ble module to write to the RX characteristic or description so I'm back to trying bluetoothSerial. (Writing the characteristic spits out a "Writing is not permitted error" and writing to the descriptor flat out crashes the Evothings workbench app.)

Any help or hints would be greatly appreciated. If this goes on much longer I'm gonna switch over to a wifi AP chip instead of BLE.

Thanks in advance for any help.

-- EF

don, 2015-03-24:
I haven't used evothing but I think it includes an old version of this plugin.

For UART, I suggest getting things working with this plugin *without* evothings workbench.

If you want more control over BLE try http://github.com/don/cordova-plugin-ble-central

shadowmouse, 2015-03-24:
Thanks for the quick response. I'll see what I can do about getting it functional outside of Evothings and post my final solution.

don, 2015-03-27:
If you continue to have a problem, re-open this issue

closed 2015-03-27 by don