don/BluetoothSerial #118: Converting serial data returned by device

rwmcintosh, 2015-03-27:
I'm not sure this is necessarily an issue for BluetoothSerial, but I can't seem to find a solution to this problem by searching myself and I hoped this group could help.
I'm using BluetoothSerial to connect to an embedded device. I'm writing the embedded code and the Android/iOS app.
On the embedded device, I respond to serial requests for data by breaking 16 bit data into bytes and pushing it out the serial port. My problem is that I cannot seem to find a way to convert the incoming raw data into sensible Javascript data.
The best success I've had is using charCodeAt:
`data[0].charCodeAt(0);`
to convert, but the problem is that this code will only convert a value that is less than 128 (works great up to 127). Anything larger than that is not converted. On the embedded side, I want to use the whole range of each byte from 0 to 0xFF.

rwmcintosh, 2015-03-27:
Follow up here. I'm doing a few things with the converted data
```javascript
console.log(data[0].charCodeAt(0));
```
and also comparing it with number
```javascript
data[0].charCodeAt(0) == 128
```

but none of this works as expected when the byte is > 127.

I am also getting my data by polling using 

```javascript
bluetoothSerial.read(function (data) {
                            console.log(parameter.id);
                            console.log(data[0]);
                            console.log(data[1].charCodeAt(0));
                            console.log(data[2].charCodeAt(0));
                            console.log(data[3].charCodeAt(0));
                            console.log(data[4]);
});
```

don, 2015-03-27:
bluetoothSerial.read returns a string so it's not good for binary data.

If you switch to [bluetoothSerial.subscribeRawData](https://github.com/don/BluetoothSerial#subscriberawdata) the plugin send you an ArrayBuffer in the callback. 

    // the success callback is called whenever data is received
    bluetoothSerial.subscribeRawData(function (data) {
        var bytes = new Uint8Array(data);
        console.log(bytes);
    }, failure);

Once you've converted your data to a Uint8Array you can access it by index

    foo = bytes[0];
    bar = bytes[1];

Depending on your data you might want to use a Uint16Array.

rwmcintosh, 2015-03-28:
Thanks Don. I'll give it a try.

closed 2015-03-28 by rwmcintosh