don/BluetoothSerial #120: Cannot write data using iOS

nicklx, 2015-04-07:
I'm using Ionic, so am also using the ngCordova plugin (for use with AngularJS). I can successfully connect to the Adafruit Bluetooth LE device and send & receive data using the "Bluefruit LE" app while running the "echoDemo" Arduino sketch as provided by Adafruit. 

When utilising my own Cordova app, I can sucessfully connect to the bluetooth device and read data coming FROM the Arduino however I cannot write data TO the Arduino. 

```
bluetoothSerial.write("testdata\n", function () {
          console.log("success");
        }, function (error) {
          console.log("error" + error);
});
```

I cannot seem to track down what the error is. There's nothing that comes up in the log in Xcode.

don, 2015-04-07:
Try re-installing the plugin from github, I think 915a65d004aaa20611d5c0f5d43b93e6a603e5b9 fixes your issues, but hasn't been released to the plugin repo yet.

    cordova plugin rm com.megster.cordova.bluetoothserial
    cordova plugin add https://github.com/don/BluetoothSerial.git

nicklx, 2015-04-07:
Perfect Thanks!

closed 2015-04-07 by nicklx