don/BluetoothSerial #121: Subscribe leaking memory?

Rlopez, 2015-04-08:
Hi!
I am calling subscribe/unsubscribe a number of times (once every second or so) and I get a behaviour where memory (related to the bound callbacks) seems to leak very quickly.

The code looks like this:

```javascript
function deviceCommand(command, callback)
{
	var result = [];
	bluetoothSerial.clear();
	bluetoothSerial.subscribe('\r\n',function (data){
		if(data == "\r\n")
		{
			bluetoothSerial.unsubscribe();
			if( typeof(callback) === typeof(Function) )
				callback(result);
			result = [];
		}
		else
		{
			result.push(data.substring(0, data.length - '\r\n'.length));
		}
	}, function(){
		flushComm();
		bluetoothSerial.connect( config.devicePorts[config.selectedDevice].address,function(){
			deviceCommand(command, callback);
		},
		flushComm);
	});
	bluetoothSerial.write(command + '\r');
}
```
Is this usage supposedly okay? I haven't seen any examples of this usage.

Also, I have done a test changing the bluetoothserial subscribe/unsubscribe/write for local fake calls and the memory leak does either dissapear or increase very, very slowly in comparison.

thanks!

don, 2015-04-08:
`subscribe` is intended to be called 1 time after `connect` and should be long running callback. Maybe try `read` or `readUntil` instead?

There may also be a memory leak. What platform are you seeing this on?

Rlopez, 2015-04-08:
I am seeing this in Android 5.0.2 on a Nexus7, communicating to a RN42, Cordova 4.3.0
My Java-foo is non-existent, but I did see something about "keep the callback alive" when calling unsubscribe that made me think it may have to do with it.
If this in the intended behaviour and/or can't be changed... what would be the better way of re-implementing my code with read/readuntil? Just call it in a tight loop until the end-of-message condition?

don, 2015-04-08:
The transmit and receive channels are independent. If you want to send a command and get a result, I think you need something like

    function deviceCommand(command, callback) {
        bluetoothSerial.write(command + '\r');
        bluetoothSerial.readUntil('\r\n', function (data) {
            var result = data.substring(0, data.length - '\r\n'.length);
            callback(result);
        });
    }

Rlopez, 2015-04-08:
I didn't quite get that.
My understanding of "readUntil" is that it'll return an empty string if there's nothing to read (which will be true if it is called just after the command is sent). So in your example the "readUntil" will execute, have an  empty buffer (assuming the BT didn't have time to reply to "command")  and return the callback immediately with "", am I wrong? :)

don, 2015-04-08:
You're correct. I was thinking readUntil waited for data before it returned.  (I'm glad you read the docs.) You'd need to set a timeout before calling readUntil, which seems very kludgy.

My recommended approach would be to subscribe for incoming data one time.  Just send commands and deal with responses as they arrive, don't try and tie the two together.

If `deviceCommand` is the behavior you want, you're going to need to implement this yourself. If you do, you should subscribe one time, accumulate data in the subscribe callback, then figure out how to connect the response to the request.

Rlopez, 2015-04-08:
Shouldn't "unsubscribe" clean the callback memory properly? I mean, even if my usage is a bit atypical it may seem like _any_ subscribe/unsubscribe pair will leak memory :(
Thanks for your help in any case, amazingly quick responses! :D

don, 2015-04-08:
Yes, `unsubscribe` should clean up memory properly! I'm going to look into that in a bit when I have more time and the right hardware.

The way you're using `subscribe` and `unsubscribe` isn't what I intended, but if you can make it work that's great. APIs are meant to be abused :)

In the mean time, maybe you can work around it with 

    function deviceCommand(command, callback) {
        bluetoothSerial.write(command + '\r');
    
        setTimeout(bluetoothSerial.readUntil, 1000, '\r\n', function (data) {
            // if data is '' you might want to call setTimeout again
            var result = data.substring(0, data.length - '\r\n'.length);
            callback(result);
        });
    }

Rlopez, 2015-04-08:
I'll do that, thanks for the tip!

don, 2015-04-08:
There's not a lot happening in my code with `unsubscribe`. See if the changes in [branch issue-121](https://github.com/don/BluetoothSerial/tree/issue-121) help. (I'm not able to run the code right now.)

I send a NO_RESULT back to the the original subscribe callback in the hope that Cordova will release the resource. Each time you call subscribe, it's creating and saving another callback, so that might be what you're seeing.

Rlopez, 2015-04-08:
What are the steps to get that on the device? I am afraid it may be out of my sphere of competence tbh :(

don, 2015-04-08:
Get a local copy of the issue-121 branch 

    git clone https://github.com/don/BluetoothSerial
    cd BluetoothSerial
    git branch --track issue-121 origin/issue-121
    git checkout issue-121
    
If you don't have git, you could [download a zip](https://github.com/don/BluetoothSerial/archive/issue-121.zip) of that branch.

Install the modified version of the plugin in your Cordova project

    cd /path/to/your/project
    cordova plugin rm com.megster.cordova.bluetoothserial
    cordova plugin add ~/BluetoothSerial

I'm assuming you cloned BluetoothSerial into ~/BluetoothSerial. Adjust the path of that last line if necessary.

Rlopez, 2015-04-09:
Finally managed to try it, works like a charm! I still have some memory leaks, but I think they are in the line of what I got when faking the BT response, amazing job! Thank you so much!
So yeah, this in trunk? :D

closed 2015-04-10 by don

Rlopez, 2015-04-23:
Hi again!
It seems like not all is good in paradise, I've been running some comparisons I wanted to share...

With a faked BT device (where bluetooth.write would return the right result right away)

[image]

Actually calling the BT device...

[image]

The thing is that the leak did improve with your changes to unsubscribe, but something else's still off.
Maybe I can test something for you and see if there's a way to improve the leak, otherwise I'll just get your timeout version in place! :)

don, 2015-05-03:
@Rlopez sorry you're still seeing problems. I'm not seeing problems my use cases of this library, so I'm can't spend time on this. If you find the source of the leak, send more more info or I'd be happy to accept a patch.

Rlopez, 2015-05-04:
That's totally fine, you've been incredibly helpful. I will send you a patch if I find out what's up!
Thanks again!

-rod

PR #122 (2015-04-10): send no_result to subscribe callback when unsubscribing
Fixes #121