don/BluetoothSerial #124: Need Some Help with RSSI

fallenfate1982, 2015-04-12:
I'm using the Adafruit nrf8001 module and your plugin on iOS 8 but I can seem to get the RSSI reading when the code detects the device. Can you let me know if there is anything extra I need to do on the Sketch side to make this work?

Your help will be appreciated!

don, 2015-04-13:
You shouldn't need to do anything on the special in the Arduino sketch.

When scanning for BLE devices on iOS using [list](https://github.com/don/BluetoothSerial#list) you should see the RSSI in the [success callback](https://github.com/don/BluetoothSerial#ios-2).

Once you're connected, you can get the RSSI with [readRSSI](https://github.com/don/BluetoothSerial#readrssi)

If that's not what you're seeing, share some code so I can try and duplicate it.

SUNYOTECH, 2015-06-12:
I use the example code in android 4.3
-------------------------
bluetoothSerial.readRSSI(
    function(rssi) {
        console.log(rssi);
    }
);
but  nothing got.

don, 2015-06-13:
readRSSI only works for iOS

closed 2015-09-18 by don