don/BluetoothSerial #127: Uint8Array is not defined

kwetzakwak, 2015-04-15:
Hi,

I'm using this plugin (very nice one by the way) to control some 4 legs robot. And it works fine, except when the device doesn't know Uint8Array. Do you think it's possible to use a "classic" array containing base64 values and send it insted of the Uint8Array ?

Devices tried:

C5303 : android 4.3 (works)
GS4mini : can't remember version (works)
GT-S5830 : android 2.3.7 (cyanogenMod 7) (doesn't work)
a sony (will get the ref. tonight) : android >4 (will get version tonight too) (doesn't work)
SGNote II : android ? (works)

Well, actually I solved the problem browsing the internet. I'm creating a pull request.

kwetzakwak, 2015-04-15:
Pull request #128 is supposed to solve that problem.

don, 2015-12-03:
Android versions older than 4.0 are no longer supported by Cordova 5.x.

If you're using Cordova 4.3.0 with Android 2.3 (Gingerbread) I suggest adding the a shim for Uint8Array into your application using [browserify](http://browserify.org/) and [typedarray](https://www.npmjs.com/package/typedarray) or based on the code in #128.

closed 2015-12-03 by don