don/BluetoothSerial #131: Write response cut off

tobbbe, 2015-04-17:
Im trying to connect to a RFID-scanner and then write a command to it. This should return a response, but the only thing I get back is "OK". Is it the BluetoothSerial-plugin that returns this? Because the response should look like this (according to the RFID-scanners docs):
__________
OK,
Model: 1861E (model name)
S/N: xxxxxxxxx (seriel number)
KNL_VER: V1.00 (kernel version)
STD_VER: V1.00 (user version)
BT_MACID: 00:do:17:a8:ff:d5 (BT mac address)
_________

If it is the plugin that just returns "OK" to tell me everything is ok. Is there a way to get the acctual response from the device? Ive tried to subscribe to \n \r etc but no luck. Ive also tried to do a 'read' in the Success callback of the write but its empty.

don, 2015-04-17:
You won't get any data in the success callback of `write`. You need to get data by calling `read` or `subscribe`. If you subscribed on \n, it makes sense you'd get OK first. Ideally you'd get other lines as they arrive. 

I'd try connecting a terminal program like [CoolTerm](http://freeware.the-meiers.org/) to the Bluetooth port  on my computer and test the commands and the response.

closed 2015-04-21 by tobbbe

don, 2015-04-21:
@tobbbe did you get this to work? What was the solution?

tobbbe, 2015-04-21:
Yea! It was my mistake. I was missing a '\r' in my command and was subscribed to \n when I should have listened for \r.
Sorry! And thanks for the quick help!