don/BluetoothSerial #132: ELM327 connection echo

sohandsouza, 2015-04-23:
Hi don,

I'm using BluetoothSerial to connect an Android phone to an ELM327 OBD dongle. I have used a Bluetooth chat app to confirm that the appropriate data is returned on sending command codes (like 010C, 010C, etc.).

However, while the bluetoothSerial plugin seemingly works fine to locate and connect to the dongle, when I send the command codes over to the dongle using bluetoothSerial.write, I only get the same command codes fed back to me from bluetoothSerial.subscribeRawData. I tried using bluetoothSerial.available to see if that matches the size of the reply stream, and it does.

I have not been able to find any known issues like this, so I was wondering if you might have encountered this, or know why it might be happening. Cheers.

sohandsouza, 2015-04-27:
Anyone else use this plugin for ELM327 communication? I've been wrestling with this issue for quite a while, and it's frustrating that everything seems to work seamlessly except for this. I wondered if it's because of the device to which I'm connecting, but I'd tried a few other OBD dongles, including high-end originals. TT_TT

kwetzakwak, 2015-04-29:
Hi,

Some devices are similar to commandline, it means that they actually send you back every letter they receive (it's the case for the device I'm using). Did you send something like '/r' or '/n', or even '/r/n' at the end of your command (if you use /r/n, it might send a empty command too)?

sohandsouza, 2015-04-29:
Yup, that's it. Thanks.

closed 2015-04-29 by sohandsouza

kauchemar, 2016-06-01:
hi sohandsouza 
i'am actually working on a cross-plateform app and i 'am trying to retrive data from my using elm 327
 but something always goes wrong when i try to send an obd  command or parse  data  i always get "undefined"  , i really need for help 
i want to know how i should send data to elm and how exactly should  read data 
thanks :)