don/BluetoothSerial #136: print image

JoseJinoria, 2015-05-06:
First of all congratulate the plugin has developed , I 'm using a small app that I am developing and it works fine for printing text , but I really need to print images , if I could direct as I can print imagens besides text. thanks

kwetzakwak, 2015-05-07:
Hi Jose,

I don't know so many things about printers, but maybe you can try sending a "low level document" to be printed ?

JoseJinoria, 2015-05-07:
Hello, thank you very much for responding.
I tried converting the image to base64 but not worth me because the plugin  assume is a string , I tried to convert a base64 encoded unit8 array (this time it a object) but I printed many other strange characters .Pardon my ignorance , when you say a low level document what you mean . Thanks other time

kwetzakwak, 2015-05-07:
What is your android version ?
Are you sending non-ASCII characters at some point ?

Actually I don't know exactly, I mean, when you send some print order to the printer, I assume that you need to send it with a special format, maybe it's a "too high level command" to print images... Not sure at all, I might tell you wrong things about that subject :/

JoseJinoria, 2015-05-07:
Hello, no , I'm just sent the picture in different formats , I think it will be that I must send some command to the printer to activate the graphical mode and print the image as a graphic . I do not know. thanks

kwetzakwak, 2015-05-07:
And, what about the version of Android you're using, and ASCII characters ?

JoseJinoria, 2015-05-07:
I'm using android 5.0.1 in note 4 ACII none that I know , I'll show below the code I'm using.

function printExample() {
	bluetoothSerial.list(function(device) {
		bluetoothSerial
				.connect(device[0].address, success, fail);
	}, function() {});}

function success() {
    bluetoothSerial.write("Text to print \n\r", printSuccess);}

kwetzakwak, 2015-05-07:
Mmmh, so that's not an android version problem.

Are you sure the first device listed by the list function is actually your printer ?
Also, according to some device, \n\r may send what you want to send, plus an extra blank line, but that's more likely not the problem...

JoseJinoria, 2015-05-07:
If the communication is perfect , the above example works fine, the problem is when instead of text need to print an image, then I prints strange characters. I try with Base64 an Unit8 Array , always prints strange characters. thakns

function success() {
bluetoothSerial.write(  image, printSuccess);}

kwetzakwak, 2015-05-07:
So that the might be the first thing I told you, maybe you need to send a document, and not just some text.
I suggest you to get closer to your printer specifications, and more generally, what type of input printers allow. So, you'll be able to send a "whole document" (or at least a header that will tell the printer you're printting a picture), instead of a single raw picture,

JoseJinoria, 2015-05-07:
Ok , I'll try , my question is as through the plugin can send a file with the data . thanks

kwetzakwak, 2015-05-12:
Since the app is using serial port, I guess it depends on the printer you have I think, maybe there is some trick as well to send a picture... I think I can't help you more :/

don, 2015-06-07:
You probably need to convert the image to a format the printer can understand. Maybe something like PCL or PostScript?

This plugin can send binary data. The write method accepts text or binary data. See the quick examples in the docs https://github.com/don/BluetoothSerial#write

closed 2015-09-18 by don

RavikumarRapolu, 2017-10-17:
function printExample() {
bluetoothSerial.list(function(device) {
bluetoothSerial
.connect(device[0].address, success, fail);
}, function() {});}

function success() {
bluetoothSerial.write("Text to print \n\r", printSuccess);}

iam using similar to above but iam unable to print sample text data. Can any one help me for that