don/BluetoothSerial #139: Serial Port and iOS

seb2peak, 2015-06-02:
Hi,

i'm trying the plugin in order to communicate with a BT 1.2  Sena ESD200 chip. 
Setting up the MAC of the BT chip make it for android. Now i'm struggling to make it works for iOS. The chip manufacture gave me as UUID :  0x1101 which is not really the format as those defined in the BLEDefines.h

Questions:
- Is there a way to use the serial port of a BT 1.2 chip with iOS?
- If we upgrade the BT chip to a BLE 4.0 (https://ack.me/products/BLE_Modules) will we be able to use the serial port? I've read that BLE 4.0 does not support the Serial Port Profile (SPP) profile anymore...

Thanks in advance for your help

don, 2015-06-07:
You need to be part of the Apple MFi program to use Bluetooth Classic on iOS. (Even if you were, this plugin wouldn't work, you'd need a different iOS implementation.)

There is no official SPP for BLE, but many manufacturers offer UART like service. That's how this plugin works for iOS. It supports the RedBearLab, Nordic, Laird, and BlueGiga "fake" SPP UART services.  Note that this plugin only supports BLE on iOS. If you need full BLE support try https://github.com/don/cordova-plugin-ble-central. If you need custom plugins for different hardware we can talk.

don, 2015-07-08:
It's possible the work @sonukapoor is doing as part of #148 could help you with Bluetooth Classic.

closed 2015-07-08 by don