don/BluetoothSerial #140: Can see device in list, cannot connect (Galaxy S5, BLE)

jptheripper, 2015-06-07:
Trying to connect to BLE from Android 5.0 (Galaxy S5).  
This is the device
https://www.tindie.com/products/AprilBrother/abbleshield-ble-40-module-ibeaconarduinoiosand/?pt=ac_prod_search
Which uses this firmware
http://wiki.aprbrother.com/wiki/Firmware/ZeroBeacon
That supports UART and AT commands

I can see the Device in the list on the phone, When I tap the ID, it requests the connection, and then after a few seconds goes to Blank, then Disconnected.

jptheripper, 2015-06-07:
With the same settings I can connect to an HC-06 just fine

don, 2015-06-07:
This plugin only support Bluetooth Classic for Android. That's why the HC-06 works fine.

For BLE on Android, try https://github.com/don/cordova-plugin-ble-central

You should be able to use [BluefruitLE example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) to get you started.You'll need to adjust the [UUIDs in the code](https://github.com/don/cordova-plugin-ble-central/blob/master/examples/bluefruitle/www/js/index.js#L36-L40). Based on [this document](http://wiki.aprbrother.com/wiki/Firmware/ZeroBeacon), you'll need to use FFF0 for the serviceUUID  and FFF1 for both the txCharacteristic and rxCharacteristic. And for [this line](https://github.com/don/cordova-plugin-ble-central/blob/master/examples/bluefruitle/www/js/index.js#L106) you probably want ble.write instead of ble.writeWithoutResponse.

jptheripper, 2015-06-07:
Thank you so much for the detailed response.  I will change plugins.

And thank you so much for the plugin itself.  This is my first mobile app,
and first venture into cordova, and first effort with bluetooth.  So I am a
bit overwhelmed, but making progress.

Jeff

closed 2015-06-07 by don

jptheripper, 2015-06-30:
Don,
 I was able to get everything working with the ble plugin! thank you.

That is, if i keep it all on one page.  I am using appgyver steroids, and i
would like different buttons on different views.  I can success connect on
one view, and i know im still connected because if i write from Arduino it
displays it

However, if i switch to a different view, and call app.write(), alerting
ble.isConnected i get "peripheral is not connected" and it fails to write.
If the same button and the same method is called from the same page as my
connection, it works, so i believe i have a scoping issue, but i have no
clue why the peripheral which is still connected, reports that it is not.

Jeff

don, 2015-06-30:
I'm not familiar with appgyver steroids. This plugin works best if your app is a single page HTML app. To change views, I show and hide divs instead of loading new pages.  If you load a new HTML page, Cordova reloads and the old references are lost. The Bluetooth peripheral is still connected to your phone, but you don't have a reference to it from your app.

jptheripper, 2015-06-30:
That explains alot.  Oh for the want of global scope.

I think i have the option to switch to modal windows instead of new html
pages.  I will see if that helps.