don/BluetoothSerial #141: IOS BLE problem.

dirtyhackers, 2015-06-08:
Hello, 
I am trying to use Amp'ed RF Technology BT43H dual mode Bluetooth it supports BLE and clasic mode (http://www.ampedrftech.com/products.php?product=BT43). I am trying to setup your chat example, on android it is running out of the box, on IOS it is not.
On the phone i can see and pair BT device easy, from the plugin when calling bluetoothSerial.list() i get an empty list (no devices) its the same if BT is paired and if not.

I have tried to set my device UUID from BLEDefines.h as adafruit and others, Then tried to add my device's UUID to same file as a new service but no luck.

Can you please help me to make it work.

don, 2015-06-10:
Android uses Bluetooth Classic so it requires pairing between your phone and the device.

iOS uses Bluetooth Low Energy, so you don't need to pair the device. iOS will require code modifications.  Start with [BLEDefines](https://github.com/don/BluetoothSerial/blob/master/src/ios/BLEDefines.h). You need to add the UUIDs for the Amp'ed UART or SPP service for Bluetooth low energy. (This is different than the UUID of your device.) Amp'ed should provide documentation for their UART service. I'd expect something like [Laird](http://www.lairdtech.com/brandworld/library/Application%20Note%20-%20Using%20VSP%20with%20smartBASIC.pdf) or [Nordic](https://learn.adafruit.com/introducing-adafruit-ble-bluetooth-low-energy-friend/uart-service).  

If you find the documentation for the service, I can point you in the right direction.

sonukapoor, 2015-07-02:
Hi Don, 

First of all thank you so much for providing the add-on. It really helped me a lot to get started. 
I have the same issue on iOS. Should we drop the BLEDDefines file into the project and redeploy?

Thanks

don, 2015-07-02:
Add your UUIDs to BLEDefines.h
https://github.com/don/BluetoothSerial/blob/6f48395a3296c48233a1722b4582c9d3f460dd38/src/ios/BLEDefines.h

Add your service to the discovery filter
https://github.com/don/BluetoothSerial/blob/6f48395a3296c48233a1722b4582c9d3f460dd38/src/ios/BLE.m#L210

Process your characteristics 
https://github.com/don/BluetoothSerial/blob/6f48395a3296c48233a1722b4582c9d3f460dd38/src/ios/BLE.m#L527

You might need a conditional here, depending if your characteristic is write or writeWithoutResponse
https://github.com/don/BluetoothSerial/blob/6f48395a3296c48233a1722b4582c9d3f460dd38/src/ios/BLE.m#L179

Once you get this working, send a pull request so it can be in the next release.

sonukapoor, 2015-07-02:
Thanks a lot for providing the steps. I will try to get this done. I am new to iOS dev, so it may take longer and I may get back to you if I need more help. I hope that is OK.

sonukapoor, 2015-07-02:
Hi Don, 

So I need to add the UUID of my iPad/iPod/iPhone device right?

don, 2015-07-02:
@sonukapoor No **not** the UUID of your device. Bluetooth has lots of UUIDs.

You need to add the UUID of the UART service that the Amp'ed device provides. There should be some Amp'ed documentation about a UART or SPP like service that they provide for their BLE devices. You could reverse engineer the service and characteristic UUIDs but it's better to get the documentation. Someone from Amp'ed should be able to provide this for you. If you get the documentation, I can help you figure out what to add.

sonukapoor, 2015-07-02:
@don I think I may have replied to the wrong thread. Basically I am new to iOS and BlueTooth. 
I am trying to connect my iPad to my bluetooth devices from TSL. It works fine on Android, but on iOS it just doesn't return a list of devices. Here is the device I am using: http://www.tsl.uk.com/products/1128-bluetooth-handheld-uhf-rfid-reader/. Please let me know if you would like me to open a new issue instead. Thanks again.

don, 2015-07-02:
@sonukapoor please create a new issue for this...

milanboric, 2015-07-02:
@don I have been in contact with Amp'ed and have read documentation which is publicly available. Product page is http://www.ampedrftech.com/products.php?product=BT43 and this one is aimed at ios http://www.ampedrftech.com/guides/BLE_ConnectionGuide.pdf. I found service UUID and it can be set manually to any value, but i did not find UUID's for RX and TX. They are not too happy to provide help i got eventually the source code of their demo app and it does not work unfortunately. 
I am a bit confused i know about device UUID (you told me i need RX/TX UUID's which i do not have) i tried already to set this device UUID. But what about app id i see in documentation that there is iAPAppID, IAPProtocolStrMain, IAPProtocolStrAlt i tried also to set app id to match with what is written to device but no luck still.
I will write more if i manage to get more information from them.

don, 2015-07-02:
@milanboric that's too bad they're not willing to work with you.

Try installing [Light Blue](https://itunes.apple.com/us/app/lightblue-bluetooth-low-energy/id557428110?mt=8) on your iOS device. Connect to the BT43H and select the service. It should list the characteristics. Ideally there is a TX and RX characteristic. Some manufacturers like ConnectBlue/uBlox use one FIFO characteristic for TX and TX and another characteristic for credits.

Post a screenshot of the service info from LightBlue here.

simoor, 2015-08-20:
Here is a printscreen what you see when scanning an Amp'ed BT43H device.

[image]

simoor, 2015-08-25:
@don is 26CC3F... the service ID I need to use? Tried it, but got no devices at all when using the list function?!

don, 2015-08-25:
@simoor This is a tricky problem. Normally I'd like to use the manufacturer's documentation about their UART service. Some dev boards are finicky and need settings just right before the BLE UART works. Working sample code from the manufacturer is also helpful. Unfortunately I think you're stuck reverse engineering this.

If there's only the one service, 26CC3FC0-6241-F5B4-5347-63A3097F6764, that's a good place to start. As for the 5 characteristics? I'm not sure what they mean. Conceptually for UART you'll have a service with 2 characteristics. One for transmit (TX) and another for receive (RX). The [Nordic UART service](https://learn.adafruit.com/introducing-adafruit-ble-bluetooth-low-energy-friend/uart-service) works like this. Other hardware like BlueGiga uses one characteristic for both transmit and receive. The BF8796F1-64F7-70B5-1E41-09BB46D79100 characteristic has read, write and notify so it could be both TX and RX? If you're unlucky it will work more like the ConnectBlue/uBlox hardware and you'll need to deal with a data credit characteristic too. See [this code](https://github.com/don/ConnectBlue/blob/master/www/js/index.js) for one approach to handling credits.

You might be better off starting with my generic [BLE plugin](https://github.com/don/cordova-plugin-ble-central/issues) rather than this plugin since you can use JavaScript to mess with the service and characteristics, rather than modifying ObjectiveC.  Once you know how the UART service works, it's easy to update this plugin to handle the characteristics.

simoor, 2015-08-26:
@don Thx, this helped me a lot! I found via another tool the Service ID "180f" but it has no characteristics. When using this ID, the device gets listed when calling the list function of your plugin and it can also be connected. But as I don't know any characteristics, I cannot communicate.

I got the source code of the demo app by Amped. As I'm not used in Objective C, I did not figure out how it works. Maybe you are better. You find the Xcode Project here: http://mab.to/IKeGOHqEn

I am interested in getting your library to work with this bluetooth module. I can send you one if you want to.

simoor, 2015-08-31:
@don @milanboric @sonukapoor 
Got it working with Amp'ed RF using this in BLEDefines.h

#define BLUEGIGA_SERVICE_UUID                         "180f"
#define BLUEGIGA_CHAR_TX_UUID                         "2a19"
#define BLUEGIGA_CHAR_RX_UUID                         "2a19"

@don maybe you can also put this in your plugin. Or you may even change the way the constants are defined to a list which can more easily be adopted.

The Chip we use is BT43H:
http://www.ampedrftech.com/products.php?product=BT43

Cheers
Simon

don, 2015-09-01:
@simoor Glad you got this working. It's really odd that they use those UUIDs, since the UUIDs are for the [Battery Service 0x180f](https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.battery_service.xml) and [Battery Level 0x2a19](https://developer.bluetooth.org/gatt/characteristics/Pages/CharacteristicViewer.aspx?u=org.bluetooth.characteristic.battery_level.xml).

If I add those UUIDs to the plugin will work for your case, but since they re-using the battery service other  non-UART, non-BT43H devices will show up during device discovery. Maybe there's a way to define the UUIDs in JavaScript.

milanboric, 2015-09-01:
@simoor Hello glad to see you got BT43H working, i tried just now to edit BLEDefines.h but i keep getting exception when running Xcode on device. I tried running chat and simple serial from don's examples.

I see now known peripherals is 1 which is great but then it has and exception :(  what am i doing wrong ?

debug output:
2015-09-01 23:07:04.091 HelloCordova[5076:907] Stopped Scanning
2015-09-01 23:07:04.093 HelloCordova[5076:907] Known peripherals : 1
2015-09-01 23:07:04.094 HelloCordova[5076:907] List of currently known peripherals :
2015-09-01 23:07:04.095 HelloCordova[5076:907] -[CBConcretePeripheral identifier]: unrecognized selector sent to instance 0x1f099690
2015-09-01 23:07:04.096 HelloCordova[5076:907] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[CBConcretePeripheral identifier]: unrecognized selector sent to instance 0x1f099690'
*** First throw call stack:
(0x3333f3e7 0x3b03a963 0x33342f31 0x3334164d 0x33299208 0x6204b 0x61f35 0x33c582fb 0x33314857 0x33314503 0x33313177 0x3328623d 0x332860c9 0x36e6533b 0x351a22b9 0x5b8e3 0x3b467b20)
libc++abi.dylib: terminate called throwing an exception

simoor, 2015-09-01:
This error is unknown to me. I created my own app using the ionic framework and just used dons plugin for the bluetooth connection...

I think you did something wrong in BLEDefines.h. You just need to change de BLUEGIGA_ constants and leave everything else in the file. If you remove the other constants, you may get an error like this

milanboric, 2015-09-02:
@simoor I managed to get it working. I was using iOS 6.1 for testing that was the problem. p.identifier.UUIDString does not run on iOS6 has to be p.UUID. Thank's for your info on amp'ed uuid's.

don, 2015-12-03:
Amp'ed reused the Battery Service for serial communication. This works but is not a good idea. I won't add this to the plugin, because that means that any device with the Battery Service will be discovered.

For now if you are using Amp'ed RF Technology BT43H, you'll need to manually modify the plugin. Edit [BLEDefines.h](https://github.com/don/BluetoothSerial/blob/master/src/ios/BLEDefines.h) and overwrite one of the existing UUIDs with data for Amp'ed.

    // hack for Amp'ed RF Technology BT43H
    #define BLUEGIGA_SERVICE_UUID "180f"
    #define BLUEGIGA_CHAR_TX_UUID "2a19"
    #define BLUEGIGA_CHAR_RX_UUID "2a19"

Future version of the plugin may allow the UUIDs to be defined in JavaScript.

closed 2015-12-03 by don

yarodevuci, 2018-05-26:
@milanboric do u still have that source code that u can share?

lizhaoxrj, 2018-08-02:
bluetoothSerial.connect          Why does the connection fail