don/BluetoothSerial #142: BluetoothSerial on ConnectBlue

JohnHigbie, 2015-06-10:
Hi again Don! I thought I remembered your saying earlier in the year that the BluetoothSerial plugin should work on Android via Classic Bluetooth. I'm trying to get my data throughput better than with the Red Bear module, and I got a module from ConnectBlue (OPS421), which uses standards: "Bluetooth v4.0 with Bluetooth low energy and Classic Bluetooth (Bluetooth Smart Ready)" 
Will I need to mod the code in order to use this? It does show up in the devices dropdown if I pair it, but I don't seem to be able to connect.
Thanks, 
John.

don, 2015-06-10:
The ublox OPS421 will work with Android and BluetoothSerial when it's configured to run in classic mode.

If you run this in low energy mode it will *not* work with BluetoothSerial. It could work, but I haven't added the code. (The ublox UART/SPP service it a bit of a PIA compared to other vendor's services.) For an example of the OPS421 with Bluetooth Low Energy and Cordova see https://github.com/don/ConnectBlue.

It's important to get the configuration correct on the board. I found it very easy to configure in ways that would make it not work.

JohnHigbie, 2015-06-11:
Thanks Don. It works like a champ. Radically better throughput than BLE!

don, 2015-06-11:
Yes. BLE is good for some stuff, but it is very slow for pushing large amounts of data.

closed 2015-06-11 by don