don/BluetoothSerial #143: why does this plugin not use BLE 4 for Android and Windows Phone

peefour, 2015-06-21:
Hi, 
I read the following in the readme:

"his plugin enables serial communication over Bluetooth. It was written for communicating between Android or iOS and an Arduino.

**Android and Windows Phone use Classic Bluetooth. iOS uses Bluetooth Low Energy.**"
1) Why does this plugin not use BLE 4 for Android and Windows Phone?
2) Can it be adapted to do so?

Thanks

don, 2015-06-21:
This plugin uses Bluetooth Classic because Serial Port Profile (SPP) Bluetooth Classic. The BLE stuff for iOS is a hack.

If you need Generic BLE support with Cordova on Android and iOS try https://github.com/don/cordova-plugin-ble-central.

peefour, 2015-06-21:
Thanks for answering so quickly !!

closed 2015-06-24 by don