don/BluetoothSerial #144: reading Pulse Oximeter with Bluetooth serial

loknathnimmagadda, 2015-06-24:
Hi, I am trying to read the data from a Pulse Oximeter using BluetoothSerial. I am not able to get the data with 
bluetoothSerial.read(app.readSuccess, app.onError);
bluetoothSerial.subscribe('\n', app.onData, app.onError); methods.

Can you please suggest.
Thanks.

don, 2015-06-24:
You'll need to figure out the specific details for your hardware. You might need to send a command before it sends you data. 

If the device is sending binary data, try bluetoothSerial.subscribeRawData.

loknathnimmagadda, 2015-06-24:
Thank you for responding don
can you please suggest what are the basic hardware details do i need to figure out. 
What is the command which i need to send to check whether the device is sending binary data.

Please help me out.

Thank you.

don, 2015-06-25:
I don't know what you need. It will be dependent on your device. Ideally you can read a manual or something that describes the protocol between the meter and the phone. More likely you won't have any info and need to reverse engineer the protocol.

loknathnimmagadda, 2015-06-30:
Hello don....i can get you. We have done the connectivity and we were able to read the device data.
Following is the code.

if (device.getName() != null && device.getName().equals(androidBluetoothDeviceName)) {
        logEvent(CMS50FW_BLUETOOTH_DEVICE_FOUND_MESSAGE);
        cms50FWDevice = device;

        bluetoothAdapter.cancelDiscovery();

              if (!connectionAlive()) {
            logEvent(ATTEMPTING_TO_CONNECT_TO_CMS50FW_MESSAGE);
            cms50FWConnectionListener.onConnectionAttemptInProgress();

            Log.v(TAG, Util.formatString(RETRIEVING_UUIDS_FROM_BLUETOOTH_DEVICE_FORMAT,
                    cms50FWDevice.getName(), cms50FWDevice.getAddress(), cms50FWDevice.getBluetoothClass()));

            // update the UUID with the one from the actual, physical device, if available
            ParcelUuid[] uuidArray = cms50FWDevice.getUuids();
            if (uuidArray != null) {
                for (int i = 0; i < uuidArray.length; i++) {
                    if (i == 0 && uuidArray.length > 0) {
                        // assume 0th uuid is the uuid for the service we want
                        bluetoothServiceUUID = uuidArray[i].getUuid();
                     
                    }
                }
            }

Now getting the socket and connect

 bluetoothSocket = cms50FWDevice.createRfcommSocketToServiceRecord(bluetoothServiceUUID);
                bluetoothSocket.connect();
                inputStream = bluetoothSocket.getInputStream();
                outputStream = bluetoothSocket.getOutputStream();
                cms50FWConnectionListener.onConnectionEstablished();
Now after establishing the connection 
androidBluetoothConnectionComponents.writeCommand(CMS50FWCommand.START_DATA);
Once connection is established to Device via Bluetooth socket the following command is issued to the Device in HexaDecimal Format
START_DATA((byte) 0xA1), 

Then we will wait for the acknowledgment from the target device.
while (androidBluetoothConnectionComponents.okToReadData) {
    cms50FWConnectionListener.onDataFrameArrived(getNextDataFrame());
}

:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::

Here when started the above process with Bluetooth Serial chat example,I am able to connect with the device using   bluetoothSerial.connect(device, app.onconnect, app.ondisconnect);
once connection is established: iam issuing the command through
bluetoothSerial.write(0xA1, success); // 0xA1 is the command to be issued here.

  var success = function () {
            message.value = "";
            messages.value += ("Us: " + text);
                  };
It went right till here:
But when trying to subscribe i never got an alert in app.onmessage function.
  // bluetoothSerial.subscribe("\n", app.onmessage, app.generateFailureFunction("Subscribe Failed"));
        bluetoothSerial.subscribeRawData(app.onmessage, app.generateFailureFunction("Subscribe Failed"));

when trying to read using : bluetoothSerial.read(function (data) {
           alert("Data is : "+data);
        }, failure);  Here data is empty.

can you please help me out with this or suggest anything positive if you can.

Thanks

don, 2015-06-30:
Send the binary data as an array of integers or use a typed array

    // array of int (or bytes)
    bluetoothSerial.write([0xA1], success, failure);

    // Typed Array
    var data = new Uint8Array(1);
    data[0] = 0xA1;
    bluetoothSerial.write(data, success, failure);

loknathnimmagadda, 2015-07-01:
when sent the binary data using typed array, success function is hit then the app gets stuck.

bluetoothSerial.write(data, success, failure);

I did not get the concept of bluetoothSerial.subscribeRawData() or bluetoothSerial.subscribe()

 bluetoothSerial.subscribe("\n", app.onmessage, app.generateFailureFunction("Subscribe Failed"));
(OR)
  bluetoothSerial.subscribeRawData(app.onmessage, app.generateFailureFunction("Subscribe Failed"));

as it never goes to app.onmessage function

don, 2015-07-01:
You're `app.onmessage` function will only get called if the plugin receives Bluetooth data. If it's not getting called, your not receiving data.

I'd expect the device doesn't like the data you're sending to it, so it's not sending you a reading.

Typically I find it best to get the serial protocol with the device worked out using [CoolTerm](http://freeware.the-meiers.org) or [PySerial](http://pyserial.sourceforge.net) on my computer. Once I know how the communication works, it's easier to write the Cordova app.

loknathnimmagadda, 2015-07-02:
Thankyou for the efforts Don..!! hope it will get solved

closed 2015-09-18 by don

killarsj, 2015-10-07:
Hello,
I am working on the same thing. i could connect the device to mobile and get data to mobile using BluetoothSerial. 
The problem is i cant understand that data which is in other format.Can anybody tell mi what format is that and how can i convert that to understandable data or numbers. Please Help.
Thank You in adv.

don, 2015-10-07:
@killarsj I think your best bet is to find protocol documentation for the device. If documentation is not available, find someone on StackOverflow that is using a similar device over Bluetooth.  Remember you can use [CoolTerm](http://freeware.the-meiers.org) or [node-serialport](https://github.com/voodootikigod/node-serialport) to work out the protocol then write your Cordova app.

killarsj, 2015-10-08:
Thank you don for your suggestion, i got the format. its in Hexadecimal ASCII CODE format(in case anyone needs..!).