don/BluetoothSerial #147: bluetoothSerial is not defined

XiaoLiuAI, 2015-07-01:
Dear author.

I tried to use cordova:com.megster.cordova.bluetoothserial under Meteor. However, the inspector always says ```bluetoothSerial is not defined```. 
I am using 
 - meteor 1.1.0.2
 - cordova:com.megster.cordova.bluetoothserial  0.4.3

Could you please help me on this issue?

don, 2015-07-01:
You're using meteor with Cordova on iOS or Android?

The global `bluetoothSerial` variable is created when Cordova (or PhoneGap) installs the plugin into your project.

XiaoLiuAI, 2015-07-02:
I tried it on the iphone browser on my macbook pro. I followed the guide of Meteor on https://github.com/meteor/meteor/wiki/Meteor-Cordova-Phonegap-integration.

It seems work with the ios simulator, so I misunderstand the way to use cordova (but the geolocation works on both browser and simulator). Do you know how to get the bluetooth data from web browser?

don, 2015-07-02:
The BluetoothSerial plugin requires iOS (or Android) in order to run. It won't work in the simulator because the simulator does not support Bluetooth. It will not run in the desktop or phone browser.

Generally you're going to need an app for Bluetooth on the phone or desktop. The browser won't have access. There are some Chrome APIs for Bluetooth https://developer.chrome.com/apps/bluetooth but I haven't used them yet.

XiaoLiuAI, 2015-07-02:
Thank you for your help!

closed 2015-07-02 by XiaoLiuAI