don/BluetoothSerial #148: Not able to list devices on iOS for TSL devices

sonukapoor, 2015-07-02:
Hi, 

I am trying to connect my iPad to my bluetooth device from TSL. It works fine on Android, but on iOS it just doesn't return a list of devices. Here is the device I am using: http://www.tsl.uk.com/products/1128-bluetooth-handheld-uhf-rfid-reader/.

Thank you

don, 2015-07-02:
[Your device](http://www.tsl.uk.com/products/1128-bluetooth-handheld-uhf-rfid-reader/) is Bluetooth Classic so it's not going to work with this plugin and iOS. 

[image]

When you're running on iOS, this plugin connects to a Bluetooth Low Energy device that has a service that looks like Bluetooth Classic SPP.

The device manufacturer might provide a SDK you can use to access the scanner on iOS using Bluetooth Classic. Maybe that's what they mean by "Apple iApp"? Generally you need to be part of the [MFi program](https://developer.apple.com/programs/mfi/) to use Bluetooth Classic on iOS.

It looks like your device can be configured as a Bluetooth Keyboard (Bluetooth HID). You might be able to use that to iOS to insert characters into the UI when you scan.

sonukapoor, 2015-07-03:
Ok. But I am also not able to see any other devices. I should be able to see the bluetooth of my Samsung Note 4, right?

sonukapoor, 2015-07-03:
@don I got it working with this one, which is a fork of your code:

https://github.com/matejkriz/BluetoothSerial/

don, 2015-07-03:
@sonukapoor I'm glad you got it working. Are you submitting your app to the AppStore? Please let me know if you get the app approved using this fork.

sonukapoor, 2015-07-03:
I will, but I am still testing it and I can see that the read is not working for me and readUntil, write and some other methods have not been implemented yet. Any chance you can help me out? It will be very appreciated.

don, 2015-07-08:
@sonukapoor I don't have time to work on this now or the hardware for testing. Feel free to post stuff here but expect slow response since I'll be on the road for the next month or so.

simoor, 2015-08-20:
@sonukapoor tried to get it working with Amp'ed too. I changed the RBL IDs in BLEDefines.h to the ones i found out via LightBlue (see https://github.com/don/BluetoothSerial/issues/141)

Was not able to get a list when using the list function. I tried both the version of @don and also the one of @matejkriz

Do you have any help for me, how did you get it working?

Thanks

closed 2015-09-18 by don

sclearion, 2015-12-17:
Same here. Using vanilla examples with an iPod 5 and a Samsung Galaxy note 5. Works GREAT Android but on ios the list() command returns nothing and does not allow you to specify the time limit. 3 seconds is nowhere near enough. Not seeing any of the 9 various bluetooth devices that the android command sees. The ios native bluetooth menu section sees everything like the droid, but the cordova application does not.

don, 2015-12-17:
@sclearion iOS filters the Bluetooth LE scan results to only show supported low energy devices. You'll see more and different devices if you scan from the iOS Bluetooth settings. This plugin does not support Bluetooth Classic on iOS.

sclearion, 2015-12-17:
So basically this "plugin does not work on with iOS" is what my boss is getting at and I think hes right. Even our BtLE devices are not showing up and we've tried on 3 different iphones and 2 ipod 5's. Seems like other people are having the same issue to. Had another dev shop try it across town and the same thing. Android works great but iOS is lacking. Nonetheless thank you for the work. We will use the Android portion and seek out a more rigorous BT soultion for our iOS

don, 2015-12-17:
@sclearion The plugin most definitely works on iOS. Lots of people are using it. IT ONLY SUPPORTS SOME HARDWARE ON iOS. See [README](https://github.com/don/BluetoothSerial#supported-platforms). iOS works with RedBearLab BLE hardware, Adafruit Bluefruit LE, Laird BL600, an BlueGiga. This plugin does *not* support Bluetooth Classic on iOS.

If you need generic Bluetooth Low Energy support try [cordova-plugin-ble-central](https://github.com/don/cordova-plugin-ble-central).

sclearion, 2015-12-17:
"The plugin most definitely works on iOS" + "IT ONLY SUPPORTS SOME HARDWARE": does not make sense especially when NONE of the 12 BTLE devices we have here are showing up. Have a nice day

simoor, 2015-12-17:
@sclearion you need to define the correct IDs or it cannot work by design of Bluetooth LE. If you configure the plugin and/or your device correctly, it will work fine. It wont list any devices if the correct IDs are missing. Had this issue too, and it was my fault. Please remember too that this plugin is used for Serial over BT
To find the IDs the LightBlue App on iOS might help you. Or you may ask your chip vendor.