don/BluetoothSerial #152: "Device connection was lost" During connect

jackaryas, 2015-07-17:
Having an issue when trying to connect to a bluetooth device, it seems to pair ok and hit the success callback almost immediately but then after a short while the error callback is called with "Device connection was lost". For reference is does never actually connect to the device ( a bluetooth speaker for testing purposes).

Am i right in thinking that the connect function should do the pairing and initiate the connection to the device?

I am using Android Kitkat and a Phillips Bluetooth BT 100 speaker for my testing. 

Any help would be greatly appreciated, thanks. 

Below is some of the code im using (ignore the stupid debugging alerts xD)

``` html
    <tr ng-repeat="d in bluetooth.devices">
         <td>{{d.name}}</td>
         <td>
            <button  ng-click="connect(d.id)" class="btn bg-light-blue pull-right">
              <i ng-if="bluetooth.connecting === d.address" class="fa fa-spinner fa-pulse"></i>
                  <i class="fa fa-wifi"></i> 
                      Connect
              </button>
            </td>
    </tr>
```

``` javascript
    function connect(addr) {
        $scope.bluetooth.connecting = addr;
        bluetoothSerial.connect(addr, function () {
            alert("Connected - Queue Life of Brian");
            status();
            $scope.bluetooth.connecting = false;
        },
                function (err) {
                    console.log(err);
                    alert("Your Rubbish");
                    status();
                    $scope.bluetooth.connecting = false;
                });
    }
```

don, 2015-07-17:
If you try and connect to an unpaired device, Android will go through the pairing process. You can pair the device before hand and connect to it.

I'd expect the speaker is disconnecting because it only supports A2DP and not RFCOMM. This plugin needs RFCOMM for sending serial data.

jackaryas, 2015-07-20:
Thanks that does appear to be the case with the speaker not supporting RFCOMM, will take that into account when purchasing the actual device in the future. 

Cheers

closed 2015-08-14 by don

nishiltamboli, 2017-06-22:
Hi @don 
I am also not able to connect to my speaker. Its a Bluetooth v2.1+EDR audio receiver. The connect method always returns error. So you think this is because of A2DP profile? You can refer to below logcat error that I am getting.

____
06-22 12:10:31.563 10360 14004 D BluetoothSerial: connect to: 11:09:33:36:6A:FE
06-22 12:10:31.564 10360 15618 E BluetoothSerial: java.io.IOException: read failed, socket might closed or timeout, read ret: -1
06-22 12:10:31.564 10360 15618 D BluetoothSerial: Sending event: error
06-22 12:10:31.564 10360 15618 I BluetoothSerial: Trying fallback...
06-22 12:10:31.565 5627 5726 E bt_btif_sock_rfcomm: btsock_rfc_signaled socket signaled for read while disconnected, slot: 6, channel: -1
06-22 12:10:31.565 10360 14004 D BluetoothSerial: setState() connecting -> connecting
06-22 12:10:31.568 10360 15618 W BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
06-22 12:10:31.570 10360 15660 D BluetoothSerial: BEGIN mConnectThread
06-22 12:10:31.571 5627 5722 W bt_l2cap: L2CA_ErtmConnectReq() PSM: 0x0003 BDA: 648af40616b2 p_ertm_info: 0x00000000 allowed:0x0 preferred:0
06-22 12:10:31.571 10360 15660 D BluetoothSerial: Connecting to socket...
06-22 12:10:31.572 5627 5722 E bt_btif : bte_scan_filt_param_cfg_evt, 1
06-22 12:10:31.574 10360 15660 W BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
06-22 12:10:31.575 5627 5726 E bt_btif_sock_rfcomm: find_rfc_slot_by_id unable to find RFCOMM slot id: 8
06-22 12:10:31.575 10360 15660 E BluetoothSerial: java.io.IOException: read failed, socket might closed or timeout, read ret: -1
06-22 12:10:31.575 10360 15660 D BluetoothSerial: Sending event: error
06-22 12:10:31.576 10360 15660 I BluetoothSerial: Trying fallback...
06-22 12:10:31.578 10360 15660 W BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
06-22 12:10:31.579 5627 5722 W bt_l2cap: L2CA_ErtmConnectReq() PSM: 0x0003 BDA: 110933366afe p_ertm_info: 0x00000000 allowed:0x0 preferred:0
06-22 12:10:31.579 5627 5722 W bt_l2cap: L2CAP - st: CLOSED evt: 21
06-22 12:10:31.579 5627 5722 W bt_l2cap: L2CAP - st: CLOSED evt: 7
06-22 12:10:31.607 5627 5666 W bt_hci_packet_fragmenter: reassemble_and_dispatch complete l2cap packet received:l2cap full length 12 and hci length 20.
06-22 12:10:31.607 5627 5722 W bt_l2cap: L2CAP - st: W4_L2CAP_CON_RSP evt: 13
06-22 12:10:31.607 5627 5722 W bt_rfcomm: PORT_StartCnf failed result:2
06-22 12:10:31.608 5627 5726 E bt_btif_sock_rfcomm: find_rfc_slot_by_id unable to find RFCOMM slot id: 9
06-22 12:10:31.608 10360 15660 E BluetoothSerial: Couldn't establish a Bluetooth connection.
06-22 12:10:31.608 10360 15660 D BluetoothSerial: Sending event: error
06-22 12:10:31.611 10360 15660 D BluetoothSerial: Sending event: connectionFailed
06-22 12:10:31.611 10360 15660 D BluetoothSerial: stop
06-22 12:10:31.611 10360 15660 D BluetoothSerial: setState() connecting -> none
06-22 12:10:33.974 5627 5722 W bt_l2cap: L2CAP - st: CLOSED evt: 1
06-22 12:10:33.974 5627 5722 W bt_sdp : SDP - Rcvd conn cnf with error: 0x4 CID 0x45
06-22 12:10:33.974 5627 5722 E bt_btif_sock_rfcomm: find_rfc_slot_by_id unable to find RFCOMM slot id: 6
06-22 12:10:33.974 5627 5722 W bt_l2cap: L2CAP - st: CLOSED evt: 1
06-22 12:10:33.975 5627 5722 W bt_rfcomm: PORT_StartCnf failed result:4
06-22 12:10:33.975 5627 5726 E bt_btif_sock_rfcomm: find_rfc_slot_by_id unable to find RFCOMM slot id: 7
06-22 12:10:33.975 10360 15618 E BluetoothSerial: Couldn't establish a Bluetooth connection.
06-22 12:10:33.976 10360 15618 D BluetoothSerial: Sending event: error
06-22 12:10:33.986 10360 15618 D BluetoothSerial: Sending event: connectionFailed
____