don/BluetoothSerial #153: No free context space; the APP has registered 20 services.

adamos42, 2015-07-23:
Hello here, I have an other problem with the connecting:
```
E/BluetoothSerialService( 7070): java.io.IOException: [JSR82] close: abort() failed.
E/BluetoothSerialService( 7070): 	at android.bluetooth.BluetoothSocket.close(BluetoothSocket.java:443)
E/BluetoothSerialService( 7070): 	at com.megster.cordova.BluetoothSerialService$ConnectThread.cancel(BluetoothSerialService.java:408)
E/BluetoothSerialService( 7070): 	at com.megster.cordova.BluetoothSerialService.start(BluetoothSerialService.java:94)
E/BluetoothSerialService( 7070): 	at com.megster.cordova.BluetoothSerialService.connectionFailed(BluetoothSerialService.java:232)
E/BluetoothSerialService( 7070): 	at com.megster.cordova.BluetoothSerialService.access$500(BluetoothSerialService.java:27)
E/BluetoothSerialService( 7070): 	at com.megster.cordova.BluetoothSerialService$ConnectThread.run(BluetoothSerialService.java:392)
```
I installed the plugin with the ```cordova plugin add com.megster.cordova.bluetoothserial``` 
Android version is ```4.2.2``` and Cordova version is ```5.1.1```

don, 2015-08-14:
I haven't see this. Do you have more than 20 bluetooth devices paired?

Sometimes the Bluetooth stack has problems and rebooting the device helps.

adamos42, 2015-08-15:
Im tryed rebooting, but the problem was not solve and I not have 20 bluetooth devices paired, just 2. The problem is in the abort() method, someway they cant close the socket.

don, 2015-08-18:
@adamos42 what phone are you using? Is the problem specific to one device? Is the problem only on 4.2.2? Can you create the simplest app that duplicates the problem? I can see if I can duplicate on my hardware.

adamos42, 2015-08-20:
@don Evolveo Strongphone Q4 with Android 4.2.2, No the problem appears not only one device but one device type, im using also a Caterpillar B15Q with Android 4.4.2 where the problem is not exists. Sot the problem is with the Android version + Cordova version, because soon as I build with 5.1.1 Cordova the problem appears.

The simpliest app: try connect to something on Android 4.2.2 and build the app with Cordova 5.1.x

On the Android 4.2.2 with Cordova 4.x im still get the errors in https://github.com/don/BluetoothSerial/issues/111 but I can handle it, so its just annoying thing.

don, 2015-12-03:
@adamos42 I can't duplicate this on a Google Galaxy Nexus running 4.2.2. with Cordova 5.1.1

Unfortunately, I think this is a hardware or Android issue with that particular phone model.

closed 2015-12-03 by don