don/BluetoothSerial #154: Build failed in Xcode 7

omnicode, 2015-08-11:
Create project "Hello world", add plugin and try to build.

../Plugins/com.megster.cordova.bluetoothserial/BluetoothSerial/CBPeripheral+Extensions.m:29:15: No visible @interface for 'CBPeripheral' declares the selector 'UUID'
../Plugins/com.megster.cordova.bluetoothserial/BluetoothSerial/CBPeripheral+Extensions.m:30:82: Property 'UUID' not found on object of type 'CBPeripheral *'

mounir01, 2015-08-12:
Hi, I have the same problem as you. did you find a solution to this?

omnicode, 2015-08-12:
Nope, just ask my friend - ios developer for help. Waiting for his answer. I he find the solution give you a notify here.

sandeepmistry, 2015-08-12:
@omnicode @mounir01 you can try changing the following lines from:

```
    NSString *uuidString = NULL;
    if ([self UUID]) {
        uuidString = (__bridge_transfer NSString *)CFUUIDCreateString(NULL, self.UUID);
    } else {
        uuidString = @"";
    }
```

to

```
    NSString *uuidString = NULL;
    if ([self identifier]) {
        uuidString = (__bridge_transfer NSString *)CFUUIDCreateString(NULL, self.identifier);
    } else {
        uuidString = @"";
    }
```

https://github.com/don/BluetoothSerial/blob/master/src/ios/CBPeripheral%2BExtensions.m#L28

```UUID``` was deprecated in iOS7, so might be removed in iOS9 ... 

* https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBPeripheral_Class/index.html#//apple_ref/occ/instp/CBPeripheral/UUID
* https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBPeripheral_Class/index.html#//apple_ref/occ/instp/CBPeripheral/identifier

mounir01, 2015-08-12:
tried this, but it's giving me this error: incompatible pointer types passing retainable parameter of type 'NSUUID*' to CF function expecting 'CFUUIDRef' (aka 'const struct_CFUUID*') type

sandeepmistry, 2015-08-12:
@mounir01 second attempt try:

```
    NSString *uuidString = self.identifier.UUIDString;
```

mounir01, 2015-08-12:
that's what i get:
[image]

sandeepmistry, 2015-08-12:
@mounir01 sorry, I mean't replace the entire ``if`` statement with that line ...

```
    NSString *uuidString = NULL;
    if ([self UUID]) {
        uuidString = (__bridge_transfer NSString *)CFUUIDCreateString(NULL, self.UUID);
    } else {
        uuidString = @"";
    }
```

becomes:

```
NSString *uuidString = self.identifier.UUIDString;
```

mounir01, 2015-08-12:
thank's lot, it works.
I stell have a similar problem with uuid in MEGBluetoothSerial. please can you help me with that.
my problem is:
[image]

sandeepmistry, 2015-08-12:
Similar change:

```
        if (p.UUID != NULL) {
            // Seriously WTF?
            CFStringRef s = CFUUIDCreateString(NULL, p.UUID);
            NSString *uuid = [NSString stringWithCString:CFStringGetCStringPtr(s, 0)
                                                encoding:(NSStringEncoding)NSUTF8StringEncoding];
            [peripheral setObject: uuid forKey: @"uuid"];
            [peripheral setObject: uuid forKey: @"id"];
        }
        else {
            [peripheral setObject: @"" forKey: @"uuid"];
        }
```

to

```
        NSString *uuid = p.identifier.UUIDString;
        [peripheral setObject: uuid forKey: @"uuid"];
        [peripheral setObject: uuid forKey: @"id"];
```

mounir01, 2015-08-12:
thank's sandeepmistry it works

omnicode, 2015-08-12:
I am still recieve strange errors

Undefined symbols for architecture x86_64:
  "_CBAdvertisementDataManufacturerDataKey", referenced from:
      -[CBPeripheral(com_megster_rfduino_extension) setAdvertisementData:RSSI:] in CBPeripheral+Extensions.o
  "_OBJC_CLASS_$_CBPeripheral", referenced from:
      l_OBJC_$_CATEGORY_CBPeripheral_$_com_megster_rfduino_extension in CBPeripheral+Extensions.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

omnicode, 2015-08-13:
Problem solved. Thanks!

don, 2015-08-14:
Thanks @sandeepmistry 

I'll get these changes rolled into a future version

don, 2015-09-18:
Fixed by #161

closed 2015-09-18 by don

halygo, 2016-02-05:
Hello @sandeepmistry 

I have a littel problem : NSString *uuid = p.identifier.UUIDString; Use  of undeclared identifier 'p' 

can you help me

Thanks