don/BluetoothSerial #155: cant list bluetooth com port

aslani75, 2015-08-14:
I am not able to list the com port near me. Am  trying like this:

            var verify = function () {
                $ionicPlatform.ready(function () {
                    $cordovaBluetoothSerial.isEnabled().then(
                            function () {
                                $scope.btBluehTooh = true;
                            },
                            function () {
                                $scope.btBluehTooh = false;
                            });
                });
            }
            $scope.list = function () {
                var promise = $cordovaBluetoothSerial.list();
                promise.then(function (data) {
                    $scope.listBT = data;
                },
                        function (error) {
                            alert(error);
                        });
            };
            $timeout(verify, 3000);

I can see that cordovaBluetoothSerial i enabled but i cant see the com port when list function is called?

don, 2015-08-14:
`list` will only show paired devices.

I suggest getting a simple example to work without ionic, then get it to work with Ionic.

aslani75, 2015-08-14:
Thx don it seemes to work with these 2 functions. First connecting insecure then sending data:

            $scope.connectBT = function () {
                $ionicPlatform.ready(function () {
                    $cordovaBluetoothSerial.connectInsecure('68:17:29:78:1F:35').then(
                            function (data) {
                                alert("Connected " + data);
                            },
                            function (data) {
                                alert("Error " + data);
                            });

                });
            }

            $scope.send = function () {
                $ionicPlatform.ready(function () {
                    $cordovaBluetoothSerial.write("hello world" + '\n').then(function (result) {
                        alert("Data sent!!" + result);
                    },
                            function (error) {
                                alert(error);
                            });
                });
            };

This is the output:

BlueCove version 2.1.1-SNAPSHOT on winsock
Address: 681729781F35
Name: xxxxxx

Server Started. Waiting for clients to connect...
Remote device address: 64A651120B87
Remote device name: HUAWEI GRA-L09
hello world

From my java server:

import java.io.BufferedReader;
import java.io.IOException;
import java.io.InputStream;
import java.io.InputStreamReader;
import java.io.OutputStream;
import java.io.OutputStreamWriter;
import java.io.PrintWriter;

import javax.bluetooth.*;
import javax.microedition.io.*;

/**
 * Class that implements an SPP Server which accepts single line of message from
 * an SPP client and sends a single line of response to the client.
 */
public class SimpleSPPServer {

    //start server
    private void startServer() throws IOException {

        //Create a UUID for SPP
        UUID uuid = new UUID("1101", true);
        //Create the servicve url
        String connectionString = "btspp://localhost:" + uuid + ";name=Sample SPP Server";

        //open server url
        StreamConnectionNotifier streamConnNotifier = (StreamConnectionNotifier) Connector.open(connectionString);

        //Wait for client connection
        System.out.println("\nServer Started. Waiting for clients to connect...");
        StreamConnection connection = streamConnNotifier.acceptAndOpen();

        RemoteDevice dev = RemoteDevice.getRemoteDevice(connection);
        System.out.println("Remote device address: " + dev.getBluetoothAddress());
        System.out.println("Remote device name: " + dev.getFriendlyName(true));

        //read string from spp client
        InputStream inStream = connection.openInputStream();
        BufferedReader bReader = new BufferedReader(new InputStreamReader(inStream));
        String lineRead = bReader.readLine();
        System.out.println(lineRead);

        //send response to spp client
        OutputStream outStream = connection.openOutputStream();
        PrintWriter pWriter = new PrintWriter(new OutputStreamWriter(outStream));
        pWriter.write("Response String from SPP Server\r\n");
        pWriter.flush();

        pWriter.close();
        streamConnNotifier.close();

    }

    public static void main(String[] args) throws IOException {

        //display local device address and name
        LocalDevice localDevice = LocalDevice.getLocalDevice();
        System.out.println("Address: " + localDevice.getBluetoothAddress());
        System.out.println("Name: " + localDevice.getFriendlyName());

        SimpleSPPServer sampleSPPServer = new SimpleSPPServer();
        sampleSPPServer.startServer();

    }
}

Just in case if anyone needs to see my sample

closed 2015-08-14 by aslani75