don/BluetoothSerial #156: Stop searching devices

AvsThiago, 2015-08-19:
Hey guys, is there some way to i stop the devices search or the plugin stops by itself?

don, 2015-08-25:
[bluetoothSerial.list](https://github.com/don/BluetoothSerial#list) stops scanning after 3 seconds on iOS.

https://github.com/don/BluetoothSerial/blob/a6bdd6c632ac8d856de2636f3f660659606a6dcd/src/ios/MEGBluetoothSerial.m#L136

closed 2015-09-18 by don

sclearion, 2015-12-17:
Can this be adjusted. The native Bluetooth discover on my ipod doesn't even take that little. Its takes like 5-7 seconds at least! 3 seconds is waaay to short.

don, 2015-12-17:
Modify the ObjectiveC to adjust the timeout https://github.com/don/BluetoothSerial/blob/a6bdd6c632ac8d856de2636f3f660659606a6dcd/src/ios/MEGBluetoothSerial.m#L136

Be sure to adjust the scan and the timer.

sclearion, 2015-12-17:
I did and it worked thank you. Still gotta figure out this list() issue