don/BluetoothSerial #158: Incorrect UUID returned when connecting to Adafruit nrf8001

mnadler, 2015-08-25:
My nrf8001 is using the standard service UUID of 6E400001-B5A3-F393-E0A9-E50E24DCCA9E and I have validated this using other devices.  When seen by bluetoothSerial.list the device's UUID is said to be 07454E19-28A7-684A-7B1D-35B9DD117B34.  I am building with XCode 6.3.2 and testing on an iPod Touch.  I have also validated that a native app running on the iPod detect the correct UUID.

Please help!

don, 2015-08-25:
You are seeing the expected behavior.

6E400001-B5A3-F393-E0A9-E50E24DCCA9E is the UUID for the Nordic UART service.

The UUID you see listed with [bluetoothSerial.list](https://github.com/don/BluetoothSerial#list) is the device identifier. On Android the device identifier is the MAC address of the hardware. iOS hides the MAC address and generates a UUID for each device.

Once you connect to your nRF8001 with iOS the plugin uses the 6E400001-B5A3-F393-E0A9-E50E24DCCA9E service for serial communication.

mnadler, 2015-08-25:
Got it thanks!

closed 2015-08-25 by mnadler