don/BluetoothSerial #159: about  bluetoothSerial.setDeviceDiscoveredListener API

shizhidi, 2015-08-26:
i use the api.
i found ----------
    setDeviceDiscoveredListener: function (notify) {
        if (typeof success != 'function')
            throw 'BluetoothSerial.setDeviceDiscoveredListener: Callback not a function'

        cordova.exec(notify, null, "BluetoothSerial", "setDeviceDiscoveredListener", []);
    }

i don't know where is the success? what's the mean?
console error------------------------
Uncaught BluetoothSerial.setDeviceDiscoveredListener: Callback not a function .
I'm crazy.

don, 2015-09-18:
[setDeviceDiscoveredListener](https://github.com/don/BluetoothSerial#setdevicediscoveredlistener) is intended to be used with [discoverUnpaired](https://github.com/don/BluetoothSerial#discoverunpaired).

discoverUnpaired only works on Android. It finds new Bluetooth devices and returns a list in success callback. Discovery can take a long time, so setDeviceDiscoveredListener is a way to get a callback as each device is discovered. You'll still get the complete list when discoverUnpaired calls success.

closed 2015-09-18 by don

sclearion, 2015-12-11:
discoverUnpaired does not fire at all for me but  discoverUnpaired works fine. Using the examples from the home page. Does it come before or after  discoverUnpaired. Can you provide an example using BOTH