don/BluetoothSerial #162: connect throw error right after disconnect

uxnow, 2015-09-21:
hello, I have a user case like this:

User can choose different devices to connect.

But when I try to connect the other one right after I disconnected from the old one, 
error throws.

```javascript

var mac1 = 'xxxx';

var mac2 = 'yyyy';

var macX; // mac address current using
function switchDevice (mac, done) {

	var connectedYes = function() {
		// check if mac changed
		// if yes nothing will be done
		// else connet
		macX === mac ? done() : bluetoothSerial.disconnect(connect);
	};
		
	var connectError = function(err) {
		// we always get err here
	};

	var connect = function() {
		macX = mac;
		bluetoothSerial.connect(mac, done, connectError);
	};

	bluetoothSerial.isConnected(connectedYes, connect);
}

```

don, 2015-09-21:
Can you run `adb logcat` and find the stack trace or more specific error messages and post the details?

If this specific to any Android version or Android hardware?

uxnow, 2015-09-22:
adb logcat

```
I/Web Console(23850): try to check bluetoothSerial isConnected:7
E/BluetoothSerialService(23850): 	at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java:872)
E/BluetoothSerialService(23850): 	at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java:96)
D/KeyguardViewMediator(  523): Received android.bluetooth.device.action.ACL_DISCONNECTED connectedBTforSmartLock = false
I/BluetoothConnectionReceiver(29419): onReceive(context, Intent { act=android.bluetooth.device.action.ACL_DISCONNECTED flg=0x8000010 cmp=com.google.android.googlequicksearchbox/com.google.android.search.core.service.BluetoothConnectionReceiver (has extras) }, [BluetoothDevice: address=00:18:E4:2D:7F:41, alias=null, name=Gprinter, majorDeviceClass=7936, deviceClass=7936])
I/Web Console(23850): try to check bluetoothSerial isConnected:7
D/KeyguardViewMediator(  523): Received android.bluetooth.device.action.ACL_CONNECTED deviceClass = 1028
I/BluetoothConnectionReceiver(29419): onReceive(context, Intent { act=android.bluetooth.device.action.ACL_CONNECTED flg=0x8000010 cmp=com.google.android.googlequicksearchbox/com.google.android.search.core.service.BluetoothConnectionReceiver (has extras) }, [BluetoothDevice: address=8C:DE:52:D9:73:9E, alias=null, name=Gprinter, majorDeviceClass=1024, deviceClass=1028])
I/Web Console(23850): try to check bluetoothSerial isConnected:7
E/BluetoothSerialService(23850): 	at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java:872)
E/BluetoothSerialService(23850): 	at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java:96)
D/KeyguardViewMediator(  523): Received android.bluetooth.device.action.ACL_DISCONNECTED connectedBTforSmartLock = false
I/BluetoothConnectionReceiver(29419): onReceive(context, Intent { act=android.bluetooth.device.action.ACL_DISCONNECTED flg=0x8000010 cmp=com.google.android.googlequicksearchbox/com.google.android.search.core.service.BluetoothConnectionReceiver (has extras) }, [BluetoothDevice: address=8C:DE:52:D9:73:9E, alias=null, name=Gprinter, majorDeviceClass=1024, deviceClass=1028])

```

uxnow, 2015-09-22:
the last lines were loged by switching to the other devices, and no response any more. thank you for so fast reply from you, @don .

uxnow, 2015-09-22:
I guess that the disconnect callback was invoked before the connection had really closed.

don, 2015-09-23:
The plugin returns success [after the service is stopped](https://github.com/don/BluetoothSerial/blob/master/src/android/com/megster/cordova/BluetoothSerial.java#L112). But I guess the actual socket takes some time to close. I don't think there's a better Android callback to hook into. 

Can you add a delay after disconnecting and before attempting the next connection?

uxnow, 2015-09-23:
I will try. thank you very much! And I suggest there should be a better way to switch between devices.

closed 2015-09-23 by uxnow

don, 2015-09-23:
@uxnow I agree adding a timeout isn't a good solution. The right way to do this could be to stay connected to both devices. See #58 for how some other people have done that running multiple versions of the plugin.

uxnow, 2015-09-23:
thank you, I will try it.