don/BluetoothSerial #164: ELM327 (OBD2)

TommyJ1994, 2015-09-27:
Hi,
I need to send commands to an ELM327 / OBD Dongle. 

Example, to get the vehicle RPM, I send 010C - and i should get a result back in hex.

I am having difficulty reading the documentation provided and I have spent hours and hours trying to get it to work.

Can you tell me which methods i should used to send a code, and then receive the reply. I am having trouble about what write, read, subscribe etc all mean.

Also if you have any good references for explaining how Bluetooth communications work at a low level, that would be great.

Thanks for the great plugin!
T.

don, 2015-10-07:
@TommyJ1994 You should be able to send and receive ODB2 data. Unfortunately, I don't know the specifics about the ODB2 protocol. 

Generally I'd send binary data to the ELM327 in an ArrayBuffer using [bluetoothSerial.write](https://github.com/don/BluetoothSerial#write). I'd get data back from the ELM327 in the success callback of  [bluetoothSerial.subscribeRawData](https://github.com/don/BluetoothSerial#subscriberawdata).

closed 2015-12-02 by don

n1705771, 2017-12-01:
Hi Tommy, could you let me know if you have solved this problem?