don/BluetoothSerial #165: polling for a bluetooth connection

pm-deepak-dhage, 2015-10-06:
Hi,

I am using this plugin to connect with HC-05 module. 
For every 4 seconds, I am checking if bluetooth is connected using bluetoothSerial.isConnected(). If it is not connected, I am doing bluetoothSerial.connect().
The reason I am doing this is, if there is a power failure and HC-05 turns off, when power comes on, app on phone should again connect to HC-05 automatically and start communication.

The issue which I am facing is, when I cut the power to HC-05, after some time, app starts to get timeout and read error for bluetooth connect request which is perfectly normal. But when I power it on again, I continue to get timeout whereas HC-05 module changes its mode to connected with device state (blinks twice every 2 sec). If this happens, my UI activity never gets a connection and I have to power off-power on HC-05 to see if it works. This happens intermittently and there is no pattern. Sometimes my app properly gets a connection (through a polling mechanism) and communicates with HC-05.
I suspect, this may be the issue with threading logic or there may be chances that HC-05 automatically gets connected with app through some android process/service which I am not aware of.

I am developing this app in cordova due to lack of knowledge about android system. Any help in this regard would be highly appreciated.

Thanks,
Deepak.

don, 2015-10-07:
Rather than polling, can you use the failure callback from [connect](https://github.com/don/BluetoothSerial#connect) to initiate the reconnection?

When the re-connection fails, are you seeing any errors in `adb logcat`? Perhaps you need to re-connect slower? Does it matter which version of the Android you're using?

pm-deepak-dhage, 2015-10-08:
Thanks for the prompt response, Don! Really appreciate it.

Your suggestion works when I am not already connected to Bluetooth through my app. What if I am connected to Bluetooth module and power goes off. I need a way to identify this and connect back when power comes on and that is why I need to check if I am connected to back-end on periodic interval. One thing I can do is, keep checking for connection and connect only when it is not connected and failure callback from previous connect has executed. This way, I will be making sure that I am not just spawning multiple connectThread in the back-end which play around shared state.

Spawning multiple threads was the reason callback sent from backend was not reaching UI as another thread was changing the shared state object before handler actually sends callback to UI.

I will keep you posted on the results.

closed 2015-12-02 by don