don/BluetoothSerial #166: Android connection with Arduino

luigibancaro, 2015-10-07:
Hello

I'm working on an application that has to connect and send a string to an Arduino device.
My app works OK in iOS (iPhone) and I have no problems communicating with Ardunio via bluetooth using your plugin.

Now I'm trying to do it for Android but I'm not able to establish a connection with my Samsung S5.
Arduino device is detected with discoverUnpaired(), I get the address but connect() and connectInsecure() fail and the failure message is: "Unable to connect to device""

Can you help me please? Thank you

don, 2015-10-07:
What Arduino Bluetooth hardware are you using?

luigibancaro, 2015-10-08:
Hi

it's based on chip nRF8001

don, 2015-10-08:
@luigibancaro the nRF8001 only supports Bluetooth Low Energy, so it won't work on Android with this plugin. (This plugin used Bluetooth Low Energy on iOS and Bluetooth Classic on Android.)

Newer Android devices support Bluetooth Low Energy and can communicate with the nRF8001. Take a look at my [BLE plugin](https://github.com/don/cordova-plugin-ble-central). One of the [example projects](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) connects to the nRF8001 using the Nordic Semiconductor SPP Service.

luigibancaro, 2015-10-08:
Thank you Don.

can it works with a Nokia Lumia, Windows Phone 8.1 ?

don, 2015-10-08:
Theoretically BLE works on WP8.1, but my plugins don't support it. Bluetooth Classic works OK on WP.

Depending on your project you could use a hardware that supports both Bluetooth Classic and BLE or for a prototype just add 2 radios.

luigibancaro, 2015-10-08:
Thank you, then https://github.com/don/cordova-plugin-ble-central could work with Windows Phone 8.1? Because I read that supports iOS and Android

luigibancaro, 2015-10-08:
It seems not

don, 2015-10-08:
@luigibancaro https://github.com/don/cordova-plugin-ble-central does not support WP8 yet. The current implementation is just a stub to keep the [PhoneGap Developer App](http://app.phonegap.com) happy. I'll be looking into WP8 (or likely Windows 10) at some point in the future when a client needs it.

closed 2015-10-29 by don