don/BluetoothSerial #167: Failure statuses

equidna, 2015-10-28:
Every time I use BluetoothSerial functions they show only the fail status independently of the function I use.
What could I be doing wrong?

The js code is:

document.addEventListener("deviceready", onDeviceReady, false);
function onDeviceReady() {
    $(function(){
        $( "#enable" ).click(function() {
            bluetoothSerial.enable(
                function() {
                    alert("Bluetooth is enabled");
                },
                function() {
                    alert("The user did *not* enable Bluetooth");
                }
            );
        });
        $( "#isEnabled" ).click(function() {
            bluetoothSerial.isEnabled(
                function() {
                    alert("Bluetooth is enabled");
                },
                function() {
                    alert("Bluetooth is *not* enabled");
                }
            );
        });
        $( "#discover" ).click(function() {
            bluetoothSerial.discoverUnpaired(function(devices) {
                alert("as");
                devices.forEach(function(device) {
                    console.log(device.id);
                })
            }, failure);
            function failure() {
                alert("abc");
            }
        });
    })
}

The HTML code is:

<html>
    <head>
        <meta charset="utf-8" />
        <meta name="format-detection" content="telephone=no" />
        <meta name="msapplication-tap-highlight" content="no" />
        
        <meta name="viewport" content="user-scalable=no, initial-scale=1, maximum-scale=1, minimum-scale=1, width=device-width, height=device-height, target-densitydpi=device-dpi" />
        <link rel="stylesheet" type="text/css" href="css/index.css" />
        <title>Hello World</title>
    </head>
    <body>
        <div class="app">
            <h1>Teste Bluetooth</h1>
            <button type="button" id="enable">enable</button>
            <button type="button" id="isEnabled">isEnabled</button>
            <button type="button" id="discover">discover</button>
        </div>
        <script type="text/javascript" src="cordova.js"></script>
        <script src="js/jquery-2.1.4.js"></script>
        <script type="text/javascript" src="js/index.js"></script>
        <script type="text/javascript">
        </script>
    </body>
</html>

And I added the following line to config.xml:

<gap:plugin name="com.megster.cordova.bluetoothserial" version="0.3.3"/>

I'm using an Android smartphone, and I'm building the app online with "phonegap build".

don, 2015-10-28:
Your code worked OK for me. Make sure you're using an up to date version of PhoneGap.

    npm install -g phonegap

I use phonegap to install the plugin, rather than editing config.xml. Note that the plugin id changed for newer versions of the plugin.

    phonegap plugin add cordova-plugin-bluetooth-serial --save    

I run compile and deploy to my phone with the command line.

    phonegap run android --device

You need to run this on a phone. Bluetooth won't work in the emulator.

equidna, 2015-10-29:
Does that means that when I install a plugin with:

phonegap plugin add ...

In the CLI I don't need to edit the config.xml?

don, 2015-10-29:
If you're using the CLI on your machine, you do *not* need to edit config.xml. Simply install the plugin with phonegap.

    phonegap plugin add cordova-plugin-bluetooth-serial

If you're using [PhoneGap Build](http://build.phonegap.com) you need to add the following to config.xml.

    <gap:plugin name="cordova-plugin-bluetooth-serial" source="npm" />

With PhoneGap build you'd build with 

    phonegap remote build android

equidna, 2015-10-29:
Looks like its not a problem with this particular plugin but with any plugin. I'm still new at phonegap.
I'm probably doing something wrong, the code works fine, except for the plugins.
Would you be so kind to direct me to some kind of tutorial or help about plugins? I know this no longer has anything to do with the issue, but I would appreciate the help.

equidna, 2015-10-30:
Found the problem. I thought phonegap and phonegap build where the same system. But the installation of plugins are made in diferent ways:

phonegap:
phonegap plugin add cordova-plugin-bluetooth-serial --save

phonegap build:
add the line "<gap:plugin name="cordova-plugin-bluetooth-serial" source="npm" />" into config.xml

closed 2015-10-30 by equidna