don/BluetoothSerial #168: Phonegap build

equidna, 2015-10-30:
When installing BluetoothSerial via phonegap build I tested the plugin using 3 API methods:
- bluetoothSerial.enable
- bluetoothSerial.isEnabled
- bluetoothSerial.discoverUnpaired

bluetoothSerial.isEnabled works fine while "enable" and "discoverUnpaired" give me an error:

TypeError: Object #<Object> has no method 'enable'
TypeError: Object #<Object> has no method 'discoverUnpaired'

Does BluetoothSerial has a newer version with those methods, that no longer support phonegap build?

don, 2015-10-30:
Older version of the plugin will have fewer methods.

You should be using version 0.6.4 of the plugin from NPM. The old version of the plugin in the PhoneGap Build repository is no longer supported. 

PhoneGap Build supports installing plugins from NPM, so there's no reason not to use the latest version of the plugin.

Add this line to config.xml

    <gap:plugin name="cordova-plugin-bluetooth-serial" source="npm" />

equidna, 2015-10-30:
When you said version 0.6.4 did you meant 0.4.4?

don, 2015-10-30:
Yes

closed 2015-10-30 by equidna