don/BluetoothSerial #169: Android to PC connection.

equidna, 2015-11-01:
Is it possible to connect the android with my PC's bluetooth?
My app seems to detect my PC but cannot establish a connection.

equidna, 2015-11-02:
I just found the problem, The PC's bluetooth's UUID is different from the UUIDs used for making the secure and insecure connections.
If I change the UUID line in BluetoothSerialService.java:
UUID.fromString("7A9C3B55-78D0-44A7-A94E-A93E3FE118CE");
to:
UUID.fromString("MY-LAPTOP-UUID");
Could I make the plugin work?

don, 2015-11-02:
The UUID 7A9C3B55-78D0-44A7-A94E-A93E3FE118CE is not used in this plugin. That is the UUID from the example code that was creating a socket to listen for connections.

The plugin connects to the SPP UUID [00001101-0000-1000-8000-00805F9B34FB](https://github.com/don/BluetoothSerial/blob/5599b8bc29096fada107ca17e00fe76cc81cd7b5/src/android/com/megster/cordova/BluetoothSerialService.java#L42).

If you are running a program on your computer that accepts Serial Port Profile (SPP) connections you should be able to connect Android to your computer.

don, 2015-11-03:
Try this to connect to your computer...

Install the [Chrome Serial Monitor](https://chrome.google.com/webstore/detail/serial-monitor/ohncdkkhephpakbbecnkclhjkmbjnmlo?utm_source=chrome-ntp-launcher) extension.

On my Mac, I open the extension and connect to /dev/cu.Bluetooth-Incoming-Port

[image]

Pair your Android phone with your computer.

Run the [Chat Example](https://github.com/don/BluetoothSerial/tree/master/examples/Chat) from the Bluetooth Serial Project

    cd examples/Chat
    cordova platform add android
    cordova plugin add cordova-plugin-bluetooth-serial
    cordova run android --device

Connect to your computer and send data

[image]

equidna, 2015-11-09:
Any advice for android to ubuntu serial connection?
Thank you.

don, 2015-11-09:
I think the [Chrome Serial Monitor](https://chrome.google.com/webstore/detail/serial-monitor/ohncdkkhephpakbbecnkclhjkmbjnmlo?utm_source=chrome-ntp-launcher) extension works on Linux.

Assuming you have Bluetooth set up correctly on your Linux box, you could use Python, Node.js, or Ruby to open a connection to the bluetooth serial port.

closed 2015-11-10 by equidna

cosmointeractive, 2017-02-08:
I am trying this but keep getting "Error: Unable to connect to device" after taping on the (already paired) listed device. Any advice?

don, 2017-02-08:
@cosmointeractive I just tried the [Chat example](https://github.com/don/BluetoothSerial/tree/master/examples/Chat) on my Nexus 5 and Nexus 5X connecting to my Mac running [Chrome Serial Monitor](https://chrome.google.com/webstore/detail/serial-monitor/ohncdkkhephpakbbecnkclhjkmbjnmlo?utm_source=chrome-ntp-launcher) on OS X 10.12.3 with Chrome 56. 

Initially I had some problems getting the Nexus 5X paired, but eventually it worked.

I'd try deleting the pairing from the computer and re-pair the phone. If possible try with a different computer and/or phone. Also restarting the Bluetooth on both devices sometimes helps.

cosmointeractive, 2017-02-09:
Thanks. I'll try that.