don/BluetoothSerial #170: Android to unknown device connection.

equidna, 2015-11-03:
This is a rather specific issue.
I want to create an android app for connecting to <a href="http://www.process-controls.com/metex-corporation/metex_pdf/hydrometer_pdf/BluetoothReceiver868_Datasheet_Eng.pdf">this</a> device. Checking an already existing app for it (and although I don't understand java) I found this line of code:

<i>UUID_SPP = UUID.fromString("00001101-0000-1000-8000-00805F9B34FB");</i>

It is the only UUID number I found on the entire project.
After I made a small test app to test the BluetoothSerial plugin, I found that I could find the MAC address of the Bluetooth device, but would get the error "Unable to connect to device" when using the "connect" and "connectInsecure" methods.

Since I'm using the correct MAC address and the UUID seems to be "00001101-0000-1000-8000-00805F9B34FB", what could be originating this behavior?

don, 2015-11-03:
If you are using this plugin with Android, it uses Bluetooth Classic. The plugin will connect to peripherals using [Serial Port Profile (SPP)](https://developer.bluetooth.org/TechnologyOverview/Pages/SPP.aspx). The code uses the well known SPP UUID 00001101-0000-1000-8000-00805F9B34FB when it's making the connection to the selected peripheral. This UUID is not related to the MAC address of your device.

If your hydrometer supports SPP, you should be able to pair your phone to the device and then connect using this plugin. If the hydrometer does not support SPP, you'll need another way to connect.

equidna, 2015-11-11:
I've managed to connect to the hydrometer and found that when I send the serial command "0x44" I get 29 bytes of data (probably the data I want to read) but when I read its just unreadable text:

D0Lڮ� ��Ap��wn��
��	)

Is it because the device is sending the information encoded with a protocol? How could I obtain readable text?

equidna, 2015-11-11:
Using a serial reader in my PC and sending the command "0x44" I've got the following information, in hex, each time:

1 -> 11 1a 19 44 30 4c da ae 9b 20 cc 01 a2 11 13 00 13 70 48 28 9d 70 c8 1e 3f ba 75 b2 b5 09
2 -> 11 1a 19 44 30 4c da ae 9b 20 cc 01 a2 21 13 00 13 70 71 59 47 8d aa 6a 55 d7 5e b5 33 0a
3 -> 11 1a 19 44 30 4c da ae 9b 20 cc 01 a2 31 13 00 13 70 66 76 0e 26 8b b9 8c 0c 47 b7 3c 09
4 -> 11 1a 19 44 30 4c da ae 9b 20 cc 01 a2 41 13 00 13 70 03 ba f2 77 6e 82 81 0d 08 ba c8 09
5 -> 11 1a 19 44 30 4c da ae 9b 20 cc 01 a2 11 13 00 13 70 48 28 9d 70 c8 1e 3f ba 75 b2 b5 09

closed 2015-11-13 by equidna