don/BluetoothSerial #175: Uncaught ReferenceError: bluetoothSerial is not defined

mmopty, 2015-11-16:
Hello everyone !

I wanted to test this little plugin to connect a phone (android or iOS) to a HC-06 but I've got some problems to install the plugin apparently.

My config is : 
- Codejs v4.2.2 (so npm in v2.14.7)
- Phonegap v5.2.0 (because it was written here (https://build.phonegap.com/plugins/1256) that it was comptible with and I didn't know for earlier versions)
- Samsung galaxy S2

What I did is that I copied the LED example in a directory and then installed the plugin using " phonegap plugin add com.megster.cordova.bluetoothserial" in this directory.
Then I run this with the Phonegap Developer App. The app opens, and I can see the button "Refresh". When I click on it I can read "Looking for Bluetooth Devices...", but that'all. No list appears...

When I use the logcat, I can read the error "E/Web Console(27120): Uncaught ReferenceError: bluetoothSerial is not defined:30". Line 30 in the LED example, is the next line after the text "Looking for Bluetooth Devices..." and uses BluetoothSerial.list(). First time in the code taht the app uses a BluetoothSerial function...

Could you help me please to solve my problem? I think that it can be a little mistake in installing the plugin, but I don't see... I'm stuck after several hours of searching...

Thank you very much and nice work (seeing the examples !)

Marc

don, 2015-11-16:
`Uncaught ReferenceError: bluetoothSerial is not defined` means that the plugin isn't installed.

I think you're seeing this since you're using the [PhoneGap Developer App](http://app.phonegap.com/). The PhoneGap Developer app only works with plugins that were installed when the app was built. Unfortunately BluetoothSerial is not included.

If you have the Android SDK installed, you can use `phonegap run` to build and install code on your phone.

If you'd rather not install the Android SDK you can use [PhoneGap Build](http://build.phonegap.com). 

With PhoneGap Build, you need to edit config.xml to install the plugin.

    <gap:plugin name="cordova-plugin-bluetooth-serial" source="npm" />

Use the command line tool to upload to Adobe's servers

    phonegap remote build android

Log into [PhoneGap Build](http://build.phonegap.com) and download the APK for your phone.

mmopty, 2015-11-17:
Thank you very much Don for your super fast answer!
I didn't know that the plugin install doesn't work with the Developer App...
Now it's working!

closed 2015-11-17 by mmopty