don/BluetoothSerial #179: Unable to build with newest versions

simoor, 2015-12-03:
hi Don

Got following error when trying to build:

.../Plugins/com.megster.cordova.bluetoothserial/BluetoothSerial/CBPeripheral+Extensions.m:29:15: error: no visible @interface for 'CBPeripheral' declares the selector 'UUID'
    if ([self UUID]) {
         ~~~~ ^~~~
.../Plugins/com.megster.cordova.bluetoothserial/BluetoothSerial/CBPeripheral+Extensions.m:30:82: error: property 'UUID' not found on object of type 'CBPeripheral *'
        uuidString = (__bridge_transfer NSString *)CFUUIDCreateString(NULL, self.UUID);
                                                                                 ^
2 errors generated.

//some more output here...

.../Plugins/com.megster.cordova.bluetoothserial/BluetoothSerial/MEGBluetoothSerial.m:69:40: error: property 'isConnected' not found on object of type 'CBPeripheral *'
        if(_bleShield.activePeripheral.isConnected)
                                       ^
.../Plugins/com.megster.cordova.bluetoothserial/BluetoothSerial/MEGBluetoothSerial.m:361:15: error: property 'UUID' not found on object of type 'CBPeripheral *'
        if (p.UUID != NULL) {
              ^
.../Plugins/com.megster.cordova.bluetoothserial/BluetoothSerial/MEGBluetoothSerial.m:363:56: error: property 'UUID' not found on object of type 'CBPeripheral *'
            CFStringRef s = CFUUIDCreateString(NULL, p.UUID);
                                                       ^
.../com.megster.cordova.bluetoothserial/BluetoothSerial/MEGBluetoothSerial.m:414:40: error: property 'isConnected' not found on object of type 'CBPeripheral *'
        if(_bleShield.activePeripheral.isConnected)

** BUILD FAILED **

(replaced my project folder with ... for easier reading.

Already removed the plugin and added it again.

i'm trying to compile on OS X Yosemite, Xcode 7.1.1

i read something that this may be deprecated or not used anymore here:
https://github.com/PunchThrough/Bean-iOS-OSX-SDK/issues/17

dou you have any idea if this problem is caused by the plugin or if its on my machine?

Thanks
Simon

don, 2015-12-03:
It looks like you have some old plugin code. With version 0.4.5 the path should be `platforms/ios/Chat/Plugins/cordova-plugin-bluetooth-serial/BluetoothSerial`

The plugin id changed recently with the move from Cordova repo to NPM. Can you try removing `com.megster.cordova.bluetoothserial` and adding `cordova-plugin-bluetooth-serial`?

simoor, 2015-12-04:
Yes, that did it. I was still using com.megster.cordova.bluetoothserial
By using cordova-plugin-bluetooth-serial it worked.
Good support!

closed 2015-12-04 by simoor