don/BluetoothSerial #181: bluetoothSerial is not defined

LazloDB, 2015-12-17:
Hi, 

as the title says, I'm having a problem that many had. 
I'm using Ionic and I've uninstalled and reinstalled the plugin multiple times.
I've also deleted cordova and reinstalled it, but that didn't fix my problem.

Do I have to load the plugin in a certain way? If so, can you tell me how?
I'm stuck on this problem for a while now and I just can't fix it.
I've also tested it on live devices and chrome://inspect gives the same errors.

Here's the code in my controller:

'use strict';
angular.module('starter')
    .controller('btCtrl', function ($scope) {
		bluetoothSerial.isEnabled(
    		function() {
       			console.log("Bluetooth is enabled");
    		},
    		function() {
       			console.log("Bluetooth is *not* enabled");
    		}
		);			
});

don, 2015-12-17:
You need to call `bluetoothSerial` after `$ionicPlatform.ready` has fired.

    ionic start btserial blank
    cd btserial
    ionic plugin add cordova-plugin-bluetooth-serial
    ionic platform add android

Edit www/js/app.js

    angular.module('starter', ['ionic'])

    .run(function($ionicPlatform) {
      $ionicPlatform.ready(function() {
        if(window.cordova && window.cordova.plugins.Keyboard) {
          // Hide the accessory bar by default (remove this to show the accessory bar above the keyboard
          // for form inputs)
          cordova.plugins.Keyboard.hideKeyboardAccessoryBar(true);

          // Don't remove this line unless you know what you are doing. It stops the viewport
          // from snapping when text inputs are focused. Ionic handles this internally for
          // a much nicer keyboard experience.
          cordova.plugins.Keyboard.disableScroll(true);
        }
        if(window.StatusBar) {
          StatusBar.styleDefault();
        }
        bluetoothSerial.isEnabled(function () {
            alert("Bluetooth is Enabled.");
          }, function (reason) {
            alert("Bluetooth is *not* Enabled.");
          }
        );
      });
    })

Run the app on your phone

    ionic run android --device

[image]

LazloDB, 2015-12-18:
It's working, 
thank you for your help and the fast reply!

closed 2015-12-18 by LazloDB

priyasaini, 2016-06-08:
HI: I want to use BluetoothSerial Plugin inside my ionic app. How should i inject it inside the controller?