don/BluetoothSerial #182: HC-05 connect (send & receive)

spidermarvel, 2015-12-27:
I am a native app in android using HC-05 bluetooth handled by a microchip 16F88. I'm doing a port with Ionic using the plug-in, I get to discover and connect but no receipt. You can use your code for any Bluetooth module (supports SPP)? thanks

spidermarvel, 2015-12-28:
Continuous, hehe. I tried to communicate the pc through hyperterminal chat example. Also, by Serial Monitor Chrome extension; in both cases it works very well.

But from my project gets nothing. The project in detail: 16F88 microcontroller (in CSS C) HC-05 + BT + distance sensor; constantly sends the distance in centimeters.

The plug-in Bluetooth Serial, how you get the data? maybe: array, numbers, text, strings, ...

Cheers

don, 2015-12-28:
The HC-05 should work fine with Android. Try the [chat example](https://github.com/don/BluetoothSerial/tree/master/examples/Chat). It looks for String data terminated with a line feed (\n). 

If you want something other than a String, try [subscribeRawData](https://github.com/don/BluetoothSerial#subscriberawdata). The callback will receive an ArrayBuffer, that can be used to create a typed array like Unit8Array or Float32Array.

spidermarvel, 2015-12-29:
Hola Don, muchas gracias por contestar.

Estuve analizando el ejemplo Chat y necesitaba simplificar aún más, sin bidireccionalidad. Por ello, estoy usando el ejemplo Counter desde el cual obtengo datos desde el PIC. 

Hi Don, thank you very much for answering.

I analyzed the example Chat and needed further simplify without bidirectional. So I'm using the Counter example from which I get data from the PIC.

Currently I have this code in the PIC, but I will change as I understood your instructions, \ n to end the line feed:

#include <16F886.h>

      #device adc=8 
      #FUSES HS, NOPUT, PROTECT, NOCPD, NOWRT, NODEBUG 
      #use delay(clock=20000000) 
      #use rs232(baud=9600,parity=N,xmit=PIN_C6,rcv=PIN_C7,bits=8,STREAM=PORT1, ERRORS)
      
      int16 distancia, tiempo;          // Defining variables 

      #define trig pin_B1     // Change as you wish, can use any pin in the MCU 
      #define echo pin_B0     // Change as you wish, can use any pin in the MCU 

void main()
{
      delay_ms(1000);                             // Boot-up delay, for troubleshooting 
      setup_timer_1(T1_INTERNAL|T1_DIV_BY_8);     // initiating timer 
  
 
   while (TRUE) 
   {
         output_high(trig);                         // ping the sonar
         delay_us(20);                            // sending 20us pulse 
         output_low(trig); 
         
         while(!input(ECHO))                       // wait for high state of echo pin 
         {} 

         set_timer1(0);                           // setting timer zero 
         
         while(input(ECHO))                       // Wait for high state of echo pin 
         {} 
         
         tiempo=get_timer1();                       // Getting the time 
         
         distancia =(tiempo*0.028) + 1.093 ;            // Calculating the distance 
                                     
                                     
         fprintf(PORT1, "\f%Lu",distancia); // Putting the time and 
                                              //distance to the LCD 
         delay_ms(1000); 
  }
}

I will put something like: fprintf (PORT1, "\ f% Lu \ r \ n", distance); or fprintf (PORT1, "\ f% lu \ n", distance); PIC code and try.

Thanks again and best regards,
Javi

spidermarvel, 2015-12-30:
it's works. Just write fprintf (PORT 1, "\ flu \ n", distance); it made begin to send data to the mobile. I have communication and one-way; Now I will put a button to interact in both directions.
Thank you very much, Don

closed 2015-12-31 by spidermarvel

rpascal, 2018-11-01:
@spidermarvel I am having trouble writing data to my PIC16f877A using HC-05 were you able to get this functionality to work?

spidermarvel, 2018-11-05:
Yes, I was able to make it work and the application works. I made it in
Ionic and it detects obstacles in the back of the car

regards,
Javi López

rpascal, 2018-11-05:
Are you able to share any of the working code for this? It would be very helpful to see an example as to how someone made it work.